tlaukkan/zigbee4java #5: Reporter interface doesn't fully describe attribute

cdjackson, 2015-02-21:
The callback from the reporter interface only provides the attribute information - not the device or cluster. This means that it's difficult to set up a single reporter callback method to receive information from different clusters (or devices).

Have I missed something here? To me it seems preferable to have the additional information available in the reporter callback so the client doesn't have to (somehow) manage this itself (with multiple callbacks I guess).

tlaukkan, 2015-02-22:
We add device and cluster identifiers to the callback method?

cdjackson, 2015-02-22:
Yes - I think this would be a useful addition if you’re happy with that?  Do you want to do that change, or shall I have a look?

tlaukkan, 2015-02-22:
Who ever gets at it first :). Let me know if you start by commenting in this thread and I will do the same. I might have a time slot in couple of hours to do some work with this issue.

tlaukkan, 2015-02-22:
Doing some work on it now.

commit 0824b20b (2015-02-22): fixed #5, added  ReportListener.receivedReport method endPointId and clusterId parameters.
commit 6e3861ac (2015-02-28): Merge branch 'master' of https://github.com/tlaukkan/zigbee4java

# By Tommi S.E. Laukkanen (11) and Chris Jackson (6)
# Via Tommi S.E. Laukkanen
* 'master' of https://github.com/tlaukkan/zigbee4java:
  Changed first free end point back to 2.
  Added more debug log on demand inspection of AF message sources.
  Added support for default sending end point containing all home automation clusters.
  Added inspection of source addresses of incoming AF messages.
  Added some additional debug message information for received AF_INCOMING_MESSAGES.
  Updated readme.
  Updated readme.
  Exposed network  [... cut]
commit 0d4158dc (2015-02-28): Merge branch 'master' into dev

* master:
  fixed #5, added  ReportListener.receivedReport method endPointId and clusterId parameters.
  Added separate maven module for generating executable ZigBee console jar.
  Added missing JavaDoc comments to ZigBeeConsole.

Conflicts:
	zigbee-api/src/main/java/org/bubblecloud/zigbee/ZigBeeConsole.java
closed 2015-02-22 by tlaukkan