tlaukkan/zigbee4java #12: Adding report listener to attribute without binding or subscribing

tlaukkan, 2015-03-01:
If binding or subscribing is already done or device is broadcasting then it should be possible to add only report listener to attribute.

presslab-us, 2015-03-05:
I'm pretty sure I'm noticing the effect of this.  It seems that some of my devices will subscribe the same endpoints numerous times.  So every time I restart the software it subscribes again and I'll get many simultaneous reports because of this.  Is there a way to query the device to check if an endpoint is already subscribed?

Or perhaps a workaround would be to unsubscribe before every subscribe.

tlaukkan, 2015-03-05:
Hi

At least I am not aware of mechanism to check whether subscription exists except for seeing if you receive reports or not. One way to solve this would be to save the information either in application or as  part of zigbee4java functionality and just implement addReportListener method without subscribing i.e. this ticket.

-Tommi

tlaukkan, 2015-03-05:
@presslab-us, if it is not too much trouble could you add links to our Wiki for zigbee devices you are using. It is still pretty hard to find good devices so this information is nice to share:

https://github.com/tlaukkan/zigbee4java/wiki

presslab-us, 2015-03-05:
Hi Tommi, I have added a few devices to the Wiki.  What is the easiest way to find if the device supports security?  I also have a furnace thermostat I am working on implementing, I will do a pull request soon.

I can see having persistence in the application for subscriptions.  But if the application is re-installed then that information is lost.  For now I have added an unsubscribe before every subscribe.  This seems to be working but I'm not sure it's the best solution.  It's certainly easy to implement.

Ryan

cdjackson, 2015-03-05:
Hi Ryan,
For some of my battery devices, they don’t respond (as I think we discussed previously) due to sleeping. It was my intention to restore the network on initialisation - but not necessarily to persist the state as such. Within the frame of OH2, we know the addresses already as this is persisted so I was going to see if we can simply use this rather than having a separate persistence…

That was my thought anyway and it’s something Tommi and I were discussing a week or so back (hence these two ‘issues’ ) but I’ve been a bit busy with work over the past week :( It’s something I want to look at but it will likely be another week as I’m travelling up to the end of next week…

Chris

commit ee08c240 (2015-03-07): Implemented listen and unlisten command to ZigBeeConsole. fixes #12
commit 96866316 (2015-03-11): Merge commit '32b116675cb0dbc72b42d195ee9492b3ed5f1515' into buildSupport

* commit '32b116675cb0dbc72b42d195ee9492b3ed5f1515':
  [maven-release-plugin] prepare for next development iteration
  [maven-release-plugin] prepare release zigbee4java-2.0.0
  Changed default maven profile to execute all projects.
  Refactored reporter addReportListener method parameter order and updated JavaDocs.
  Implemented serialization of network state to ZigBeeApi and  ZigBeeConsole. Enables instantaneous startup of console after first startup which includes initial network browsing. Network state is serialized [... cut]
closed 2015-03-07 by tlaukkan