tlaukkan/zigbee4java #15: Everyone please join discussion group

tlaukkan, 2015-03-07:
@chris-hatton @cdjackson @presslab-us 

The URL to the FAQ: https://groups.google.com/d/forum/zigbee4java

chris-hatton, 2015-03-07:
Joined.  FYI I’m slowly working on full Gradle support for zigbee4java (with equivalent to Maven profiles for Desktop/Android). With Gradle becoming the de-facto build system for Android, this will make zigbee4java useful to a wider audience.  I am new to Gradle so it’s taking a bit of time to get right (Maven seems a lot more straightforward at the moment…).

Thanks,
Chris

tlaukkan, 2015-03-07:
Hi

Nice work Chris. There are lots of people and project appreciating gradle
support in server / desktop projects as well.

Cheers,
Tommi

tlaukkan, 2016-01-04:
By the way there is 2015 release thread at the google group. Contributors please check it out :).

closed 2016-06-16 by tlaukkan