tlaukkan/zigbee4java #17: Permit join defaults to on

presslab-us, 2015-03-16:
The permit join defaults to on, allowing any device to join the network at any time.  This should default to off, and only turned on when the user allows permit join, and only for a limited period of time.

tlaukkan, 2015-03-21:
Good point, do you have suggestion how we should expose this in API?

presslab-us, 2015-03-22:
There is the existing method `sendPermitJoinRequest` but it is not called from anything in zigbee4java.  The firmware already supports the join timeout also.

I'm thinking of adding wrappers for this method in `ZigBeeApi` like `public void permitJoin(short timeout)` and `public void permitJoin(short timeout, ZToolAddress16 address)`.  This would allow all devices to join or just one device specified by address.  The user application would call this method when joining is desired.

Also another change would be to call `sendPermitJoinRequest` to turn off join, on every startup, because the dongle defaults to on.

tlaukkan, 2015-03-22:
Sounds like a good plan. We could maybe add constructor parameter to API for configuring whether permit join should be disabled on startup? At least in lab environment it might be nice to have devices joining automatically.

Lets make the method signature use int instead of short and define the unit. This is pretty normal convention:

public void permitJoin(int timeoutMillis)
public void permitJoin(int timeoutMillis, ZToolAddress16 address)

Would you like to implement this change?

presslab-us, 2015-03-22:
Maybe use a `System.getProperty` for the permit join override, this way it can easily be set on the command line.

The firmware API for `ZDO_MGMT_PERMIT_JOIN_REQ` specifies the timeout in seconds as follows: `Specifies the duration to permit joining. 0 = join disabled. 0xff = join enabled.  0x01-0xfe = number of seconds to permit joining.`  So on the one hand I see using milliseconds as other methods use this unit,  on the other hand the firmware does not support it so precision is implicitly lost if the `permitJoin` method does the `timeoutMillis / 1000`.  I do agree with the naming convention.

Yes I'd be happy to do this change.  For these smaller changes would you like me to create a branch and let you merge it into master?

tlaukkan, 2015-03-23:
Hi

System properties are configuration mechanism for  programs and as such could be used for console but I would prefer optional command line argument. On API level it is good style to use programmatic non JVM wide way to set this kind of options. If there are security concerns that weight against API level option, I would consider leaving the option out all together rather than implementing environment variable.

As constructor parameter count is increasing for ZigBeeApi class we could:

1) create ZigBeeApiConfiguration class,
2) add another constructor with optional arguments,
3) or create setter for this option which could be invoked before startup.

In my opinion this kind of change can implemented in master as long as there is agreement regarding API level changes.

d1g2w3, 2015-03-23:
Hi Tommi

Personally I prefer a parameter parser that creates a static class of
options. Keeps it simple, removes specific dependencies and supports an
unknown amount of parameters increase.

Kind regards,

Danny Watson

presslab-us, 2015-03-24:
@tlaukkan How about something like 3) but it could be invoked after startup if desired.  On startup the join will be disabled, but it is always possible to re-set the join to always enabled with no timeout.  This way the API is simple and all that is needed is to do something like `permitJoin(true)` after startup to set it back enabled indefinitely.  The firmware supports this with the 0xff value.

The implementation in the console app could be done a number of ways.  I think at least adding another  console command to change the join setting would be good.  I don't have a strong opinion about this.

tlaukkan, 2015-03-24:
Hi

Good points and viable plan. We can either  allow permit join timeout value -1 equaling infinite timeout  or separate method with boolean parameter?

From my perspective this feature is go for direct master implementation. 👍

Best regards,
Tommi

tlaukkan, 2015-03-24:
Hi Danny

Do you refer to console parameter parsing? Console has room for improvement as it started off as an example and has been evolving since to become a development tool.

Or how to parametrize ZigBeeApi programmatically?

Br,
Tommi

chris-hatton, 2015-03-24:
To enhancing the functionality of command-line parsing at launch, I would recommend looking at Apache Commons CLI, which I've worked with in the past and provides a great framework to avoid reinventing the wheel.  https://commons.apache.org/proper/commons-cli/

For the console itself; that's a conversation to be had... I would just mention that I am currently refactoring code between **zigbee-console-common** and **zigbee-console-javase** in preparation for implementation of **zigbee-console-android**.

Summary of the changes is:
- Console is constructed with generic InputStream / OutputStream.
- Console runs on it's own Thread.
Neither of these changes should affect the current behaviour.

d1g2w3, 2015-03-24:
Hi,

The CLI and using a common library that is understood is a great idea.

On the command parser we are all of the agreement that it's upto the
application to decide what parameters to use so the parser can be issued as
an example.

I personally don't like the idea of a never ending list of parameters in
the constructor as it will never end.

Can i ask a question, why is there a need to change the permit joining from
default in the api? Can we not just leave it upto the application developer
to change this after startup?

Kind regards,

Danny Watson

presslab-us, 2015-03-24:
@d1g2w3 With permit join on, it's possible for someone malicious to compromise the security.  Once joined they are on the network; they can sniff data and control devices.  The default should err on the side of more security.

This is why the window of time for joining should be limited, and even the allowed device address can be specified.

Ryan

d1g2w3, 2015-03-24:
Hi Ryan,

Hmm, ok, they can sniff data as long as they have the HA key and see the
transport key but understand regards to controlling devices.

Kind regards,

Danny Watson

presslab-us, 2015-03-24:
@d1g2w3 The network key is only exposed when a device joins.  Otherwise it is a secret that all devices use for the cypher.

Ryan

d1g2w3, 2015-03-24:
@presslab-us Accidently said transport key instead of network key.

If you have the transport key and you are sniffing the channel you can see
the network key being passed hence you can sniff the network

presslab-us, 2015-03-26:
I've pushed my changes to master.  For the console app I've just made it have the previous functionality (permit join always enabled).  I know @chris-hatton is refactoring it right now so I didn't want to change too much.

I think the console app should eventually have some way to re-enable permit join from the command line (otherwise leaving it disabled), and a dedicated console command to enable/disable permit join and set the timeout duration.

tlaukkan, 2015-03-27:
Nice! :+1:

closed 2015-04-19 by tlaukkan