tlaukkan/zigbee4java #18: Bug in ZigBeeEndpointImpl

alcastelo, 2015-03-17:
Hi there.
A couple of days ago, I started to play with this great project and a netvox zigbee device.
I found what seems to be a little bug in the bindTo method of the ZigBeeEndpointImpl class.
When you are building the ZDO_BIND_REQ packet, the last parameter is declared as BindDstEP , but you are sending is endpoint.getDeviceTypeId().  
I think this is a  bug, as it should be sending endpoint.getEndPointAddress().
I will try to send a patch for your consideration.
Thanks you very much for your time and work.
Regards!!!!

tlaukkan, 2015-03-21:
@alcastelo this looks like a valid fix. Would you like to make a pull request?

alcastelo, 2015-03-21:
Thanks a lot.
I´m making a pull request right now!!!

tlaukkan, 2015-03-21:
Thank you :+1:

closed 2015-03-21 by tlaukkan

alcastelo, 2015-03-21:
Thanks to you for your work!! :+1: