tlaukkan/zigbee4java #19: Gradle support ready for final test and merge

chris-hatton, 2015-03-19:
Gradle support is now greatly improved from the first attempt and, in my judgement, ready to be merged from **gradleSupport** branch to **master**.  As a temporary feature branch, **gradleSupport** will be deleted, following the merge.

The current changes from **master** have been merged into **gradleSupport**, so there should be no conflict when merging back.

I do not have a suitable testing ZigBee environment at the moment, and while there is no reason why the addition of new build support should have broken anything internal to the library, it may be a good idea for someone (particularly @tlaukkan, @cdjackson  or @presslab-us) to find a moment to compile and run ZigBeeConsole etc. from this branch and perform some kind of sanity check.

As a brief intro: Gradle builds both *desktop* and *android* related artifacts by default.
See this by executing the following line from the **zigbee4java** folder:

'''
./gradlew build
'''

Which profiles are built can be controlled in the file gradle.properties.  It is not currently possible to override this property with -P on the Gradle command line.  Enabling property override behaviour seems to be a subject of some discussion within the Gradle community.

I've also modified the Maven POMs to match the new module layout and build scheme, with a 'desktop' and 'android' profile.

One major difference between the Maven and Gradle outputs, currently, is that Maven does not produce an Android APK file from the ZigBeeConsole App project.  This seems a low priority fix however as Gradle is the clear recommended system for Android builds these days.

The zigbee-console-android project is a placeholder at the moment, producing an empty Android App, but now with the correct dependencies in place to build from zigbee-console-common and construct an Android console to demo ZigBee functionality from the mobile platform... this will be my next endeavour.

chris-hatton, 2015-03-19:
I've just made a breaking change to **gradleSupport** branch again:  By changing the **zigbee-console-android** project to a full Android Studio one, a dependency on an Android Support library has been introduced which is not currently resolved by the Maven build.  Better hold on the above until this is resolved.

chris-hatton, 2015-03-20:
After further attempts, I'm calling defeat on the Maven Android build for now.
Both **android** and **desktop** compile with Gradle.
Only **desktop** compiles with Maven.
I have updated the *default* profiles in both build system configurations to match.
I hope to return to this but Maven build support for Android doesn't seem very important until someone needs it.  I would like to progress with more important areas like developing the Android Console App and verifying new firmware functionality.
@tlaukkan Please consider the merge from **buildSupport** to **master** ready for review, if you agree with priorities above.

tlaukkan, 2015-03-21:
I will try and take a look at this one today and get it merged asap.

tlaukkan, 2015-03-21:
@chris-hatton I have merged and tested gradleSupport branch. Nice work :+1:

tlaukkan, 2015-03-22:
@chris-hatton is all done with this issue from your perspective? From my side we can close this one.

chris-hatton, 2015-03-22:
Yes, with the one caveat that Maven does not currently build the **zigbee-console-android** project, all else is complete.  Let's close it.

closed 2015-03-22 by chris-hatton