tlaukkan/zigbee4java #23: Crash with Door Lock device

smulikHakipod, 2015-03-27:
```
> .22:44:21 857  ERROR  Error constructing response packet {}
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
	at org.bubblecloud.zigbee.network.packet.zdo.ZDO_MSG_CB_INCOMING.translate(ZDO_MSG_CB_INCOMING.java:145)
	at org.bubblecloud.zigbee.network.packet.ZToolPacketStream.parsePayload(ZToolPacketStream.java:309)
	at org.bubblecloud.zigbee.network.packet.ZToolPacketStream.parsePacket(ZToolPacketStream.java:171)
	at org.bubblecloud.zigbee.network.packet.ZToolPacketParser.run(ZToolPacketParser.java:107)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 6
	at org.bubblecloud.zigbee.network.packet.zdo.ZDO_SIMPLE_DESC_RSP.<init>(ZDO_SIMPLE_DESC_RSP.java:96)```

The frame is too small for the parser. Any ideas what can cause it so I can start fixing it?
Thanks

presslab-us, 2015-03-27:
I have seen something like this while developing a ZigBee device.  I believe the problem is that zigbee4java does not check if the descriptor length is zero and tries to parse it.  I think I can come up with a patch, give me a few hours.

presslab-us, 2015-03-27:
Yes, I see the problem in the code.  By the way, what lock are you working with?

smulikHakipod, 2015-03-27:
Actually I am using the sample lock from TI HA examples.

presslab-us, 2015-03-27:
Well, there you go.  If you look at `sampleDoorLock_TestEp` in the sample code, you can see there is no simple descriptor.  This is what triggers the bug in zigbee4java.  You can just remove the `afRegister` for this endpoint, you don't need it.

What sort of lock are you working on?  I plan to make some kind of DIY ZigBee servo actuator for my deadbolt.

presslab-us, 2015-03-27:
This should fix it, please test and report back. https://github.com/tlaukkan/zigbee4java/commit/a9654de27bcf297e82f849ed9e9a1b47151cf56b

smulikHakipod, 2015-03-27:
Thanks for your help!!
About the door lock, the manufacture of doors here uses some unique mechanism that isn't compatible with any smart lock I found. I draw using OpenSCAD a plastic that wraps the around the key and I want the CC2530 to control it.

http://imgur.com/a/dlIDZ

presslab-us, 2015-03-27:
That is a strong looking gear motor, and your part looks very nice.  Here are some of my DIY ZigBee efforts: https://groups.google.com/forum/#!topic/zigbee4java/nREaHiQW42s

smulikHakipod, 2015-03-28:
Your solution fixed the bug, many thanks!
As for the door lock, it looks like the zigbee4java does not implement the door lock device, so ZigbeeApi matches my door lock as Dimmable Light (working on fixing it now by adding DoorLockDevice?).

presslab-us, 2015-03-28:
Yes I have seen this too.  If the device ID is not supported it seemingly picks a different one at random.  Your device ID is 0x000A, yes?

smulikHakipod, 2015-03-28:
I added the changes to my fork:
https://github.com/tlaukkan/zigbee4java/compare/master...smulikHakipod:master
For now, really basic lock command and lock state.
Are the names and structure OK? Did I miss anything? Should I make a pull request?
thanks

tlaukkan, 2015-04-02:
Hi

When you are happy with your code please make a pull request and we will review and merge it to master.

Best regards,
Tommi

tlaukkan, 2015-04-02:
Nice work by the way and very useful.

tlaukkan, 2015-04-19:
Thank you for the fix presslab-us. :+1:

closed 2015-04-19 by tlaukkan