tlaukkan/zigbee4java #25: Console port speed

cdjackson, 2015-04-11:
I notice that with the refactored console, the COM port speed is now running at 38400 where it was previously running at 115k2. I've noticed a lot of errors in the past day or so - I initially put this down to the new firmware (I updated to the latest stack) but I'm not wondering if it's related to the speed. If I run my application (which runs at 115) it works ok until I start the console - then bad stuff starts (timeouts, various network addressing errors, failure to reset the stick). I've only tried this a couple of times, but it seems if I change the console to 115200 again, this goes away.
@chris-hatton - I'm guessing that you refactored this (but I've not checked). Was this a conscious change or did it 'creep' in?

This might not be the cause of my problem, but for sure the console speed has changed - just not sure if it's right?

Cheers
Chris

tlaukkan, 2015-04-18:
I suggest switching back to 115200 in master branch if this has not been done yet. Has anyone checked the TI ZStack specification related to serial port speed?

Switching to different port speeds once the dongle connectivity has been established may cause problems as well. In other words if your application first uses 115200 and then you switch it off and start console with 38400 thiscould cause problems.

cdjackson, 2015-04-18:
> Switching to different port speeds once the dongle connectivity has been established may cause problems as well. In other words if your application first uses 115200 and then you switch it off and start console with 38400 thiscould cause problems.
> 
I just tried resetting the stick and starting at 115k and I still get errors. I’ll check the spec for the port speed - it’s strange as I often get errors just resetting the stick when the api starts...

cdjackson, 2015-04-18:
I don’t think the stack spec is going to specify this (and I can’t find it there either). What I did find is in the 2531 EM manual it says -:
The port settings should be 38400 baud, 8 bits, 1 stop bit, no parity. Hardware handshaking must be enabled.
 
However, I’ve tried various incarnations of hardware handshaking (RTS/CTS anyway) and I still get errors…

tlaukkan, 2015-04-19:
I close this issue because timeouts have different issue open and they do not seem to be serial port speed related.

closed 2015-04-19 by tlaukkan