tlaukkan/zigbee4java #26: deviceUpdated callback called after every inspection

cdjackson, 2015-04-12:
The deviceUpdated callback is called after every inspection of a node, even if the node hasn't been 'updated' (unless it's a new node).

In my mind, it shouldn't call deviceUpdated unless something has changed. Currently the stack checks to see if the address has changed, and if it has it does a re-inspection of the endpoints - this is when I think it should call deviceUpdated.

Do others agree? I'll make a PR for this and you can decide :)

chris-hatton, 2015-04-12:
Sounds reasonable to me Chris... if nothing's changed then it hasn't 
'updated'.

Re: Your message about baud rate / console behaviour.

Firstly, you're right: committing 38400 as the default baud rate was 
unintentional on my part, so please let's revert it to what we know 
works i.e. 115200.  Apologies for this.

On the other hand, I can't immediately see why 38400 would cause the 
errors described.  If the CC2531/firmware couldn't handle that baud rate 
you'd expect it to not work at all?  I also experienced errors like you 
described but thought they were due to other factors.  Does anyone know 
why using a lower baud rate would cause issues? Hopefully the alternate 
baud rate isn't revealing some fundamental timing errors in the library.

Thanks,
Chris

cdjackson, 2015-04-12:
Hey Chris,

> Firstly, you're right: committing 38400 as the default baud rate was 
> unintentional on my part, so please let's revert it to what we know 
> works i.e. 115200. Apologies for this.

No probs - I’m not sure it makes a huge difference actually…

> On the other hand, I can't immediately see why 38400 would cause the 

No - I don’t think its the cause. Sometimes it works fine, other times I get various errors that I can’t pinpoint. The variable nature means just when you think you’ve found the cause, you find you haven’t :)

This morning I tried various options - old firmware, new firmware, 2.0.3 snapshot and 2.0.2, and I get at least the occasional error with versions, but it seems worse with the latter version. I’ve also just installed Parallels VM to run Windoze on my Mac (so I can program the sticks without having to dust off an old PC) so that’s another possible contender…

I’ll keep playing on that one, but maybe we should start another issue where we can list some of these things to look for commonality...

tlaukkan, 2015-04-19:
Good improvement :+1:

closed 2015-04-19 by tlaukkan