tlaukkan/zigbee4java #32: Invoking commands from ClusterListener thread cause timeout

smulikHakipod, 2015-04-18:
Hey,
I noticed something strange, if I listen for cluster messages, process them and invoking command in response I get timeout issues. If I invoke the response on the main thread instead of the ClusterListener thread its working fine. Debugging the issue I saw that zigbee4java got the response.. its just some threading issue. Am I missing something? Anyone have an Idea how to fix the issue?

cdjackson, 2015-04-18:
Interesting - do you think this might be related to #31?

I'm seeing a number of strange issues, so it might be all related.

smulikHakipod, 2015-04-18:
Further investigation shows that every thread other than ZToolPacketParser thread (the cluster listener thread) is working fine.

tlaukkan, 2015-04-18:
Hi

I suspect that invoking any ZigBee stack request response messaging from ZToolPacketParser thread will cause deadlock as your ZToolPacketParser thread will be waiting for response instead of parsing the response packet.

Best regards,
Tommi

smulikHakipod, 2015-04-22:
I put the notifyClusterListener function in a new thread and everything is working fine.
https://github.com/tlaukkan/zigbee4java/commit/9007e30af979b0c750c5be3d0f26dbd339c15691#diff-4
(just focus on ZigBeeEndpointImpl.java, I pushed the other files by accident)

closed 2016-03-30 by tlaukkan