tlaukkan/zigbee4java #51: Unable to send cluster on the ZigBee network due to: Z_NWK_NO_ROUTE

cdjackson, 2015-07-19:
Anyone know how to reverse this?

While this doesn't appear to happen with all battery devices, I have a Develco occupancy sensor - I can communicate with this fine initially, but if I do a request after it stops listening, I get a timeout (expected) and subsequently I get the NO ROUTE error - even when I put the sensor back into identify mode...

Any thoughts appreciated...

Brhett, 2015-08-03:
I am not sure whether I met the same issue as Chris did. The issue happend on reading any attributes via console. e.g. "read 0 0 0"

Failed to read attribute.
org.bubblecloud.zigbee.api.cluster.impl.api.core.ZigBeeClusterException: Read Attribute 0 failed due to:  org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerException: Unable to send cluster on the ZigBee network due to: Z_NWK_NO_ROUTE(205) (null)
	at org.bubblecloud.zigbee.api.cluster.impl.core.AttributeImpl.doClusterWideRead(AttributeImpl.java:176)
	at org.bubblecloud.zigbee.api.cluster.impl.core.AttributeImpl.getValue(AttributeImpl.java:106)
	at org.bubblecloud.zigbee.ZigBeeConsole$ReadCommand.process(ZigBeeConsole.java:1046)
	at org.bubblecloud.zigbee.ZigBeeConsole.executeCommand(ZigBeeConsole.java:217)
	at org.bubblecloud.zigbee.ZigBeeConsole.processInputLine(ZigBeeConsole.java:192)
	at org.bubblecloud.zigbee.ZigBeeConsole.start(ZigBeeConsole.java:168)
	at org.bubblecloud.zigbee.ZigBeeConsoleJavaSE.main(ZigBeeConsoleJavaSE.java:34)

Brhett, 2015-08-03:
After merging latest code b8122b8 to original f8e051b, reading attribute failed caused by different reason.
Failed to read attribute.
org.bubblecloud.zigbee.api.cluster.impl.api.core.ZigBeeClusterException: Read Attribute 0 failed due to:  org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerException: Unable to send cluster on the ZigBee network due to: Z_MAC_NO_ACK(233) (null)
	at org.bubblecloud.zigbee.api.cluster.impl.core.AttributeImpl.doClusterWideRead(AttributeImpl.java:183)
	at org.bubblecloud.zigbee.api.cluster.impl.core.AttributeImpl.getValue(AttributeImpl.java:106)
	at org.bubblecloud.zigbee.ZigBeeConsole$ReadCommand.process(ZigBeeConsole.java:1096)
	at org.bubblecloud.zigbee.ZigBeeConsole.executeCommand(ZigBeeConsole.java:217)
	at org.bubblecloud.zigbee.ZigBeeConsole.processInputLine(ZigBeeConsole.java:192)
	at org.bubblecloud.zigbee.ZigBeeConsole.start(ZigBeeConsole.java:168)
	at org.bubblecloud.zigbee.ZigBeeConsoleJavaSE.main(ZigBeeConsoleJavaSE.java:34)

drugo72, 2015-09-07:
Do you have any Wireshark capture to try to understand what's happening?

cdealti, 2015-09-07:
Do you have a Wireshark capture to try to understand what is happening?

tlaukkan, 2016-03-30:
I got Z_MAC_NO_ACK error when I had device in the network.json saved state but not actually in the ZigBee network. Better log messages at least would be nice improvement.

tlaukkan, 2016-04-01:
Confirmed that at least one reason for this is that device is read from network safe file but does not actually exist in network. As result AF_DATA_CONFIRM returns error and reports status Z_MAC_NO_ACK:

<-- AF_DATA_CONFIRM (AF_DATA_CONFIRM{Endpoint=2, Status=Z_NWK_NO_ROUTE(205), TransID=36})
Failed to read attribute.
> org.bubblecloud.zigbee.api.cluster.impl.api.core.ZigBeeClusterException: Read Attribute 0 failed due to:  org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerException: Unable to send cluster on the ZigBee network due to: Z_NWK_NO_ROUTE(205) (null)

This is intended behavior as far as I can see. Closing this issue for now. Please reopen if you find contrary evidence or reasoning.

closed 2016-04-01 by tlaukkan