tlaukkan/zigbee4java #57: Adding gnu/posix style command line options to ZigBeeConsoleJavaSE

cdealti, 2015-09-07:
I'd like to modify the console to support standard gnu/posix options like in the below example.
In order to support this I need to slightly modify the ZigBeeConsole, ZigBeeApi and ZigBeeNetworkManagerImpl to allow for more parameters to be passed to the constructors.
In order to avoid constructors with so many parameters I was thinking to the Builder pattern.
Do you think this is worthwhile to add?

Cheers,
  Cristiano

```
usage: java -jar zigbee4java-serialPort.jar [OPTION]... DEVICE
 -c,--channel <channel>                 add the specified channel (11-26)
                                        to the list of channels scanned on
                                        establishing the network. The
                                        default channel list is 11, 14,
                                        15, 19, 20, 24, 25. This option
                                        may be repeated multiple times.
 -d,--distribute-network-key            distribute the network key in
                                        clear to devices joining the
                                        network. If not specified the
                                        default behaviour is to not
                                        distribute the network key.
 -e,--epid <epid>                       the extended PAN ID (EPID) in
                                        hexadecimal format
                                        (0-0xfffffffffffffffe). The
                                        default value is 0 which means
                                        that stack will set the EPID to
                                        the value of the IEEE address.
 -k,--network-key <network-key>         the network key in hexadecimal
                                        format (1-0xfffffffffffffffe). If
                                        not set the dongle will use a
                                        preconfigured value.
 -m,--discovery-mode <discovery-mode>   the method used to discover
                                        devices. Valid values are
                                        `announce', `addressing', `lqi'
                                        and `all' as a shortcut for the
                                        first tree. This option can be
                                        repeated multiple times.
 -p,--panid <panid>                     the PAN ID in hexadecimal format
                                        (0-0xffff). The default PAN ID is
                                        0xffff which means that stack will
                                        generate a random, nonconflicting
                                        PAN ID.
 -r,--reset-network                     start with a clean network state
                                        and configuration. If not
                                        specified the network will resume
                                        using the last state and
                                        configuration.
 -s,--no-security                       disable security.
```

cdjackson, 2015-09-08:
For me, this all looks good. I just wonder if it is best to use a builder notation, or a config dictionary.

cdealti, 2015-09-08:
I'm wondering the same thing. I've started with the builder but I don't like it. The above options and possibly more need to be passed from the outermost class to the innermost one. Some options are consumed by classes in between. With Builder we need to create a parallel hierarchy of builders one for every class so that adding a new option might require to change every builder in the hierarchy. A dictionary might be simpler especially if the above configuration is read from a file. However, simply passing the dictionary to every constructor in the hierarchy is not that safe. You don't know who is using what configuration.