tlaukkan/zigbee4java #64: Handle of ZDO_SIMPLE_DESC_REQ

Brhett, 2015-09-18:
1 *ZDO_SIMPLE_DESC_REQ* would get 2 *ZDO_SIMPLE_DESC_RSP*. (I don't see why/where the 2nd *ZDO_SIMPLE_DESC_RSP* is sent. **Any ideas?**)
If a node has 2 endpoints, the result would be unexpected.

See example below. The coordinator has 2 endpoints. Inspecting endpoint 1 would take the 2nd result of inspection of endpoint 2 as its result, which is incorrect.

It could be resolved by updating the *WaitForCommand* by adding an extra parameter "endpoint". If the asynchronous response is *ZDO_SIMPLE_DESC_RSP*, checking the value of "Endpoint" field of the response to determine whether the response is dedicated to current waiter.
**Any suggestion?**

```
13:19:00 940  DEBUG  Found 2 end points on #0.
13:19:00 941  DEBUG  <-- ZDO_ACTIVE_EP_RSP (ZDO_ACTIVE_EP_RSP{ActiveEndpointCount=2, ActiveEndpointList=[2, 1], nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0)})
13:19:00 943  DEBUG  Inspecting node #0 (00:12:4B:00:01:4E:93:2C) / end point 2.
13:19:00 945  DEBUG  <- ZDO_MGMT_LQI_REQ_SRSP (ZDO_MGMT_LQI_REQ_SRSP{Status=SUCCESS(0)})
13:19:00 945  DEBUG  -> ZDO_SIMPLE_DESC_REQ (Packet: length = 5, apiId = 0x25 0x04, full data = 0xfe 0x05 0x25 0x04 0x00 0x00 0x00 0x00 0x02 0x26, checksum = 0x26, error = false, errorMessage = null) 
13:19:00 954  DEBUG  <-- ZDO_MGMT_LQI_RSP (ZDO_MGMT_LQI_RSP{NeighborLQICount=1, NeighborLQIEntries=1, NeighborLqiList=[org.bubblecloud.zigbee.network.packet.zdo.ZDO_MGMT_LQI_RSP$NeighborLqiListItemClass@53c88e], SrcAddress=0x00 0x00, StartIndex=0, Status=SUCCESS(0)})
13:19:00 954  DEBUG  Received expected response: ZDO_MGMT_LQI_RSP
13:19:00 954  DEBUG  Found 1 neighbors on node #0
13:19:00 954  DEBUG  Node #40442 visible from node #0 with LQI value 170
13:19:00 955  DEBUG  <-- ZDO_MGMT_LQI_RSP (ZDO_MGMT_LQI_RSP{NeighborLQICount=1, NeighborLQIEntries=1, NeighborLqiList=[org.bubblecloud.zigbee.network.packet.zdo.ZDO_MGMT_LQI_RSP$NeighborLqiListItemClass@f78a5], SrcAddress=0x00 0x00, StartIndex=0, Status=SUCCESS(0)})
13:19:00 959  DEBUG  <- ZDO_SIMPLE_DESC_REQ_SRSP (ZDO_SIMPLE_DESC_REQ_SRSP{Status=SUCCESS(0)})
13:19:00 959  DEBUG  Waiting for ZDO Simple Description Response
13:19:00 959  DEBUG  -> ZDO_IEEE_ADDR_REQ (Packet: length = 4, apiId = 0x25 0x01, full data = 0xfe 0x04 0x25 0x01 0xfa 0x9d 0x00 0x00 0x47, checksum = 0x47, error = false, errorMessage = null) 
13:19:00 968  DEBUG  <-- ZDO_SIMPLE_DESC_RSP (ZDO_SIMPLE_DESC_RSP{DevID=0x0000, DevVer=0, InClusterCount=0, InClusterList=[], OutClusterCount=13, OutClusterList=[0x0000, 0x0001, 0x0002, 0x0003, 0x0004, 0x0005, 0x0006, 0x0007, 0x0008, 0x0009, 0x000f, 0x000a, 0x0502], ProfID=0x0104, Endpoint=2, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=34, inputs=null, outputs=null})
13:19:00 968  DEBUG  Received expected response: ZDO_SIMPLE_DESC_RSP
13:19:00 968  DEBUG  ZDO Simple Description Response ZDO_SIMPLE_DESC_RSP{DevID=0x0000, DevVer=0, InClusterCount=0, InClusterList=[], OutClusterCount=13, OutClusterList=[0x0000, 0x0001, 0x0002, 0x0003, 0x0004, 0x0005, 0x0006, 0x0007, 0x0008, 0x0009, 0x000f, 0x000a, 0x0502], ProfID=0x0104, Endpoint=2, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=34, inputs=null, outputs=null}
13:19:00 969  DEBUG  Sender end point 00:12:4B:00:01:4E:93:2C/2 found with profile PROFILE_ID_HOME_AUTOMATION: 260
13:19:00 969  DEBUG  networkAddr: 0, inputClusters: 0
13:19:00 969  DEBUG  Inspecting node #0 (00:12:4B:00:01:4E:93:2C) / end point 1.
13:19:00 971  DEBUG  <-- ZDO_SIMPLE_DESC_RSP (ZDO_SIMPLE_DESC_RSP{DevID=0x0000, DevVer=0, InClusterCount=0, InClusterList=[], OutClusterCount=13, OutClusterList=[0x0000, 0x0001, 0x0002, 0x0003, 0x0004, 0x0005, 0x0006, 0x0007, 0x0008, 0x0009, 0x000f, 0x000a, 0x0502], ProfID=0x0104, Endpoint=2, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=34, inputs=null, outputs=null})
13:19:00 975  DEBUG  Received expected response: ZDO_SIMPLE_DESC_RSP
13:19:00 976  DEBUG  <- ZDO_IEEE_ADDR_REQ_SRSP (ZDO_IEEE_ADDR_REQ_SRSP{Status=SUCCESS(0)})
13:19:00 977  DEBUG  -> ZDO_SIMPLE_DESC_REQ (Packet: length = 5, apiId = 0x25 0x04, full data = 0xfe 0x05 0x25 0x04 0x00 0x00 0x00 0x00 0x01 0x25, checksum = 0x25, error = false, errorMessage = null) 
13:19:00 985  DEBUG  <- ZDO_SIMPLE_DESC_REQ_SRSP (ZDO_SIMPLE_DESC_REQ_SRSP{Status=SUCCESS(0)})

13:19:00 985  DEBUG  Waiting for ZDO Simple Description Response
13:19:00 985  DEBUG  ZDO Simple Description Response ZDO_SIMPLE_DESC_RSP{DevID=0x0000, DevVer=0, InClusterCount=0, InClusterList=[], OutClusterCount=13, OutClusterList=[0x0000, 0x0001, 0x0002, 0x0003, 0x0004, 0x0005, 0x0006, 0x0007, 0x0008, 0x0009, 0x000f, 0x000a, 0x0502], ProfID=0x0104, Endpoint=2, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=34, inputs=null, outputs=null}

13:19:00 985  DEBUG  Sender end point 00:12:4B:00:01:4E:93:2C/1 found with profile PROFILE_ID_HOME_AUTOMATION: 260
13:19:00 985  DEBUG  13 Adding <profileId,clusterId> <260,0> to sender2EndPoint hashtable
13:19:00 985  DEBUG  13 Adding <profileId,clusterId> <260,1> to sender2EndPoint hashtable
13:19:00 985  DEBUG  13 Adding <profileId,clusterId> <260,2> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,3> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,4> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,5> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,6> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,7> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,8> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,9> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,10> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,15> to sender2EndPoint hashtable
13:19:00 986  DEBUG  13 Adding <profileId,clusterId> <260,1282> to sender2EndPoint hashtable
13:19:00 987  DEBUG  networkAddr: 0, inputClusters: 0
Node discovered: 00:12:4B:00:01:4E:93:2C (#0)
13:19:00 987  DEBUG  Endpoint inspection completed, next inspection slot in 0
13:19:00 987  DEBUG  Inspection queue: New queue size: 2. Failed queue size: 0
13:19:00 987  DEBUG  Inspecting device 00:12:4B:00:01:4E:93:2C.
13:19:00 987  DEBUG  Inspecting existing node #0 (00:12:4B:00:01:4E:93:2C)
13:19:00 987  DEBUG  Endpoint inspection completed, next inspection slot in 100
13:19:00 994  DEBUG  <-- ZDO_SIMPLE_DESC_RSP (ZDO_SIMPLE_DESC_RSP{DevID=0x0400, DevVer=0, InClusterCount=3, InClusterList=[0x0501, 0x0003, 0x0001], OutClusterCount=3, OutClusterList=[0x0502, 0x0500, 0x0003], ProfID=0x0104, Endpoint=1, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=20, inputs=null, outputs=null})
13:19:00 995  DEBUG  <-- ZDO_SIMPLE_DESC_RSP (ZDO_SIMPLE_DESC_RSP{DevID=0x0400, DevVer=0, InClusterCount=3, InClusterList=[0x0501, 0x0003, 0x0001], OutClusterCount=3, OutClusterList=[0x0502, 0x0500, 0x0003], ProfID=0x0104, Endpoint=1, nwkAddr=0x00 0x00, SrcAddress=0x00 0x00, Status=SUCCESS(0), len=20, inputs=null, outputs=null})
13:19:01 087  DEBUG  Inspection queue: New queue size: 1. Failed queue size: 0
13:19:01 087  DEBUG  Inspecting device 00:12:4B:00:01:4E:93:2C.
13:19:01 087  DEBUG  Inspecting existing node #0 (00:12:4B:00:01:4E:93:2C)
13:19:01 088  WARN   The device #0 (00:12:4B:00:01:4E:93:2C) has been found again with a new network address #40442 
Node updated: 00:12:4B:00:01:4E:93:2C (#40442)
13:19:01 088  DEBUG  Endpoint inspection completed, next inspection slot in 99
13:19:01 187  DEBUG  Inspection queue: New queue size: 0. Failed queue size: 0
There are 0 known devices in the network.
ZigBee console ready.
```

cdealti, 2015-09-19:
Hi Brhett. You have two nodes in you network, the coordinator dongle with address 0 and another node with address 40442, correct?

Your log starts at 940 stating that there are 2 endopints on the dongle. Is this finding related to the following line at 941?
I've not looked into the code yet and the order of printouts might not be the order of events.

At 943 the discovery of the dongle's endopint 2 is started requesting it's simple descriptor but there are other things moving. For example, you receive ZDO_MGMT_LQI_RSP at 945. This probably matches a previous request not shown in the log. This response shows the existence of your other node with address 40442. However, 
**The same ZDO_MGMT_LQI_RSP is received at 955**. Is this expected?

Also note that while the discovery of the dongle's endpoint 2, the discovery process requests the IEEE address of the other node with address 40442. This requests goes over the air and the response is not showed in the log.

Then we get to the problem. The simple descriptor for the dongle's ep 2 is received at 968 and the discovery process continues with ep 1.
**However the simple descriptor for the dongle's ep 2 is received also at 971 and is understood as the simple descriptor of ep1**.

Also note that the simple descriptor of ep 1 is reported twice at 994 and 995 (but nobody is waiting for it at that point).

Is the above picture correct?

It looks like all asynchronous responses are duplicated but is this a firmware or an application problem? Can you reproduce this sistematically or did it happen just once?

About your proposal to try to match the ep number, I'd go further also matching the destination address and the network address of interest. These  parameters are common to every ZDO request.

Ciao,
  Cristiano

Brhett, 2015-09-20:
> You have two nodes in you network, the coordinator dongle with address 0 and another node with address 40442, correct?
Actually, there is only one node, the coordinator dongle. It would be discovered as  address 0 and address 40442 since I've set a custom endpoint to the dongle to make it work as a IAS CIE.
So that 2 endpoints are found. ep 1 is IAS CIE while ep 2 is the default ZigBee controller (See ApplicationFrameworkLayer).
Then, inspecting of each endpoint would be started since 943.
However, the response of simple descriptor request would be received twice at 968 and 975. As a result, the response at 975 would be incorrectly considered as the result of inspection of ep 1 since 969.

***So the problem is that why there are two response of simple descriptor request for each endpoint?***

"destination address" and "network address" can not be used for resolving above issue as these two endpoints have the same values. So I have to use the "endpoint" provided by the znp interface to handle this case.

cdealti, 2015-09-20:
Hi Brhett,
  The ZigBee coordinator always has the single network address 0x0000 independently of the number of endpoints registered.
It looks like you have a second node (40442) associated to the coordinator (or it was associated in the past, you turned it off and/or you started zigbee4java without clearing the network state. Please see my note above about the IEEE address request sent to this node.

About the suggestion to match ep, destination network address e address of interest, it does not help in this specific case but in those cases where you could receive a delayed response from a node while you are inspecting another one. It's just defensive code.

Can you reproduce the problem systematically?

Ciao,
   Cristiano

Brhett, 2015-09-21:
Hi Cristiano,

You are right. The node 40442 is another node, which was associated in the past. It disappear after resetting the dongle (4th parameter is set to "true" for clearing the network state).

> Can you reproduce the problem systematically?

Yes. It could be reproduced easily. Add codes below in start() of ZigBeeConsole besides https://github.com/Brhett/zigbee4java/commit/bb3428b39793a4b9ac247569def24dd46837581a 
```java
ZigBeeNetworkManager manager = zigbeeApi.getZigBeeNetworkManager();
manager.addCustomEndpoint("1", "260", "1024", "0", "[0,1,3,1281]", "[3,1280,1282]");
```

Regards,
  Brhett

cdealti, 2015-09-21:
Hi Brhett,

This leads us to your other issue https://github.com/tlaukkan/zigbee4java/issues/63.
Is it possible that this API was deprecated due to the above issue? I cannot test it right now but what happens if you add another custom ep?

Ciao,
  Cristiano

Brhett, 2015-09-21:
Hi Cristiano,

> Is it possible that this API was deprecated due to the above issue?

I don't know. I found the API has been marked as deprecated since the 1st revision of ZigBeeNetworkManager.java (copying from z4o?)

> I cannot test it right now but what happens if you add another custom ep?

If we don't resolve the duplicated response of *ZDO_SIMPLE_DESC_RSP*, the "list" command would not list the added custom virtual device (IAS CIE in this case).

Regards,
  Brhett

cdealti, 2015-09-22:
Hi Brhett,
  I've just fetched your custom_device branch and it works fine for me.
I'm testing on Ubuntu 12.04 and an Oracle JDK 7 starting the console like this:

```
java -jar target/zigbee-console-javase-2.0.3-SNAPSHOT.jar /dev/ttyACM0 11 0x1234 true
```

The relevant part of the log is:

```
09:32:56 657  DEBUG  -> AF_REGISTER (Packet: length = 41, apiId = 0x24 0x00, full data = 0xfe 0x29 0x24 0x00 0x01 0x04 0x01 0x00 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x01 0x00 0x02 0x00 0x03 0x00 0x04 0x00 0x05 0x00 0x06 0x00 0x07 0x00 0x08 0x00 0x09 0x00 0x0f 0x00 0x0a 0x00 0x0c 0x00 0x15 0x00 0x02 0x07 0x00 0x01 0x00, checksum = 0x00, error = false, errorMessage = null) 
09:32:56 670  DEBUG  <- AF_REGISTER_SRSP (AF_REGISTER_SRSP{Status=SUCCESS(0)})
09:32:56 670  INFO   Registered default sending endpoint 1 with clusters: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 15, 10, 12, 21, 1794, 256]
09:32:56 670  DEBUG  Adding <profileId,clusterId> <260,0> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,1> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,2> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,3> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,4> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,5> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,6> to sender2EndPoint hashtable
09:32:56 671  DEBUG  Adding <profileId,clusterId> <260,7> to sender2EndPoint hashtable
09:32:56 672  DEBUG  Adding <profileId,clusterId> <260,8> to sender2EndPoint hashtable
09:32:56 672  DEBUG  Adding <profileId,clusterId> <260,9> to sender2EndPoint hashtable
09:32:56 672  DEBUG  Adding <profileId,clusterId> <260,15> to sender2EndPoint hashtable
09:32:56 673  DEBUG  Adding <profileId,clusterId> <260,10> to sender2EndPoint hashtable
09:32:56 673  DEBUG  Adding <profileId,clusterId> <260,12> to sender2EndPoint hashtable
09:32:56 673  DEBUG  Adding <profileId,clusterId> <260,21> to sender2EndPoint hashtable
09:32:56 673  DEBUG  Adding <profileId,clusterId> <260,1794> to sender2EndPoint hashtable
09:32:56 674  DEBUG  Adding <profileId,clusterId> <260,256> to sender2EndPoint hashtable
09:32:56 675  DEBUG  -> AF_REGISTER (Packet: length = 41, apiId = 0x24 0x00, full data = 0xfe 0x29 0x24 0x00 0x02 0x04 0x01 0x00 0x00 0x00 0x00 0x00 0x10 0x01 0x01 0x02 0x01 0x00 0x02 0x01 0x02 0x02 0x02 0x04 0x02 0x00 0x03 0x03 0x04 0x00 0x04 0x01 0x04 0x02 0x04 0x04 0x04 0x05 0x04 0x06 0x04 0x04 0x0b 0x00 0x05 0x14, checksum = 0x14, error = false, errorMessage = null) 
09:32:56 688  DEBUG  <- AF_REGISTER_SRSP (AF_REGISTER_SRSP{Status=SUCCESS(0)})
09:32:56 688  INFO   Registered default sending endpoint 2 with clusters: [257, 258, 512, 513, 514, 516, 768, 1027, 1024, 1025, 1026, 1028, 1029, 1030, 2820, 1280]
09:32:56 688  DEBUG  Adding <profileId,clusterId> <260,257> to sender2EndPoint hashtable
09:32:56 688  DEBUG  Adding <profileId,clusterId> <260,258> to sender2EndPoint hashtable
09:32:56 689  DEBUG  Adding <profileId,clusterId> <260,512> to sender2EndPoint hashtable
09:32:56 689  DEBUG  Adding <profileId,clusterId> <260,513> to sender2EndPoint hashtable
09:32:56 689  DEBUG  Adding <profileId,clusterId> <260,514> to sender2EndPoint hashtable
09:32:56 690  DEBUG  Adding <profileId,clusterId> <260,516> to sender2EndPoint hashtable
09:32:56 690  DEBUG  Adding <profileId,clusterId> <260,768> to sender2EndPoint hashtable
09:32:56 690  DEBUG  Adding <profileId,clusterId> <260,1027> to sender2EndPoint hashtable
09:32:56 690  DEBUG  Adding <profileId,clusterId> <260,1024> to sender2EndPoint hashtable
09:32:56 691  DEBUG  Adding <profileId,clusterId> <260,1025> to sender2EndPoint hashtable
09:32:56 691  DEBUG  Adding <profileId,clusterId> <260,1026> to sender2EndPoint hashtable
09:32:56 691  DEBUG  Adding <profileId,clusterId> <260,1028> to sender2EndPoint hashtable
09:32:56 691  DEBUG  Adding <profileId,clusterId> <260,1029> to sender2EndPoint hashtable
09:32:56 692  DEBUG  Adding <profileId,clusterId> <260,1030> to sender2EndPoint hashtable
09:32:56 692  DEBUG  Adding <profileId,clusterId> <260,2820> to sender2EndPoint hashtable
09:32:56 692  DEBUG  Adding <profileId,clusterId> <260,1280> to sender2EndPoint hashtable
09:32:56 693  DEBUG  -> AF_REGISTER (Packet: length = 13, apiId = 0x24 0x00, full data = 0xfe 0x0d 0x24 0x00 0x03 0x04 0x01 0x00 0x00 0x00 0x00 0x00 0x02 0x01 0x05 0x02 0x05 0x2e, checksum = 0x2e, error = false, errorMessage = null) 
09:32:56 700  DEBUG  <- AF_REGISTER_SRSP (AF_REGISTER_SRSP{Status=SUCCESS(0)})
09:32:56 700  INFO   Registered default sending endpoint 3 with clusters: [1281, 1282]
09:32:56 700  DEBUG  Adding <profileId,clusterId> <260,1281> to sender2EndPoint hashtable
09:32:56 700  DEBUG  Adding <profileId,clusterId> <260,1282> to sender2EndPoint hashtable
09:32:56 701  DEBUG  Sending permit join with data: 0
09:32:56 702  DEBUG  -> ZDO_MGMT_PERMIT_JOIN_REQ (Packet: length = 5, apiId = 0x25 0x36, full data = 0xfe 0x05 0x25 0x36 0x0f 0xfc 0xff 0x00 0x01 0x1b, checksum = 0x1b, error = false, errorMessage = null) 
09:32:56 714  DEBUG  <- ZDO_MGMT_PERMIT_JOIN_REQ_SRSP (ZDO_MGMT_PERMIT_JOIN_REQ_SRSP{Status=SUCCESS(0)})
09:32:56 714  DEBUG  -> ZDO_MGMT_PERMIT_JOIN_REQ (Packet: length = 5, apiId = 0x25 0x36, full data = 0xfe 0x05 0x25 0x36 0x02 0x00 0x00 0x00 0x01 0x15, checksum = 0x15, error = false, errorMessage = null) 
[... cut ...]
Node discovered: 00:12:4B:00:01:48:F1:20 (#0)
09:32:56 895  DEBUG  Endpoint inspection completed, next inspection slot in 0
09:32:56 895  DEBUG  Inspection queue: New queue size: 1. Failed queue size: 0
09:32:56 895  DEBUG  Inspecting device 00:12:4B:00:01:48:F1:20.
09:32:56 895  DEBUG  Inspecting existing node #0 (00:12:4B:00:01:48:F1:20)
09:32:56 895  DEBUG  Endpoint inspection completed, next inspection slot in 100
09:32:56 995  DEBUG  Inspection queue: New queue size: 0. Failed queue size: 0
```

Brhett, 2015-09-22:
Hi Cristiano,

Have you add codes in *start* method of class *ZigBeeConsole* ?
```java
        //discoveryModes.remove(DiscoveryMode.LinkQuality);
        final ZigBeeApi zigbeeApi = new ZigBeeApi(port, pan, channel, resetNetwork, discoveryModes);

+      // create virtual device "IAS CIE"
+      ZigBeeNetworkManager manager = zigbeeApi.getZigBeeNetworkManager();
+      manager.addCustomEndpoint("1", "260", "1024", "0", "[0,1,3,1281]", "[3,1280,1282]");
```

And "list" console command expects the device "IAS Control and Indicating Equipment" is displayed if everything is fine.

Regards,
 Brhett

cdealti, 2015-09-22:
Hi Brhett,
  No I did not add your endpoint but look!!!

From my log you can see that endpoint 1 is already registered (at 09:32:56 670).
This is done by ApplicationFrameworkLayer.createDefaultSendingEndPoint().
**Your code adds a duplicate endpoint 1!!!**

I will check the createDefaultSendingEndPoint() tomorrow but there is something wrong with createDefaultSendingEndPoint(): **three default endpoints are created!**

Ciao,
  Cristiano

Brhett, 2015-09-23:
Hi Cristiano,

My custom endpoint would be registered as ***endpoint 1***. Then ***ApplicationFrameworkLayer.createDefaultSendingEndPoint()*** would create endpoints at free endpoints(i.e. starts from 2). So there would be no duplicate endpoint.

> I will check the createDefaultSendingEndPoint() tomorrow but there is something wrong with createDefaultSendingEndPoint(): three default endpoints are created!

It looks like ***AF_REGISTER*** only accepts 16 input/output clusters so that the ***createDefaultSendingEndPoint()*** has to split the output clusters into 3 different endpoints.

Regards,
Brhett

cdealti, 2015-09-23:
Hi Brhett,
   You are right, your custom endpoint would be registered first as ep 1 and the default one(s) will be assigned ep 2 etc.

I've added your code for the IAS CIE but is still works for me (see log below).

Maybe some differences in the dongle firmware version? I've added a method to get the dongle firmware version in https://github.com/cdealti/zigbee4java/commit/f63112854602be041b839280840d081557c99b59.

I got this for mine:

15:19:26 610  INFO   Dongle product ID: 0
15:19:26 611  INFO   Dongle protocol version 2
15:19:26 611  INFO   Dongle software version 2.6.3

> It looks like AF_REGISTER only accepts 16 input/output clusters so that the
createDefaultSendingEndPoint() has to split the output clusters into 3 different endpoints.

This is not correct and needs to be fixed.

```
cristiano@vespone:~/dev/git/zigbee4java/zigbee-console-javase$ java -jar target/zigbee-console-javase-2.0.3-SNAPSHOT.jar /dev/ttyACM0 11 0x1234 true
ZigBee API starting up...13:56:24 771  DEBUG  Opening portIdentifier '/dev/ttyACM0'.
13:56:24 781  DEBUG  -> SYS_RESET (Packet: length = 1, apiId = 0x41 0x00, full data = 0xfe 0x01 0x41 0x00 0x01 0x41, checksum = 0x41, error = false, errorMessage = null) 
13:56:24 803  DEBUG  <-- SYS_RESET_RESPONSE (Packet: length = 6, apiId = 0x41 0x80, full data = 0xfe 0x06 0x41 0x80 0x00 0x02 0x00 0x02 0x06 0x03 0xc2, checksum = 0xc2, error = false, errorMessage = null)
13:56:24 804  DEBUG  Resetting network stack.
13:56:24 804  INFO   Setting clean state.
13:56:24 805  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 3, apiId = 0x26 0x05, full data = 0xfe 0x03 0x26 0x05 0x03 0x01 0x03 0x21, checksum = 0x21, error = false, errorMessage = null) 
13:56:24 814  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:24 821  DEBUG  Changing the Network Mode to Coordinator.
13:56:24 822  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 3, apiId = 0x26 0x05, full data = 0xfe 0x03 0x26 0x05 0x87 0x01 0x00 0xa6, checksum = 0xa6, error = false, errorMessage = null) 
13:56:24 831  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:24 831  INFO   Resetting dongle.
13:56:24 832  DEBUG  -> SYS_RESET (Packet: length = 1, apiId = 0x41 0x00, full data = 0xfe 0x01 0x41 0x00 0x01 0x41, checksum = 0x41, error = false, errorMessage = null) 
13:56:26 717  DEBUG  <-- SYS_RESET_RESPONSE (Packet: length = 6, apiId = 0x41 0x80, full data = 0xfe 0x06 0x41 0x80 0x00 0x02 0x00 0x02 0x06 0x03 0xc2, checksum = 0xc2, error = false, errorMessage = null)
13:56:26 719  DEBUG  Setting channel to 11.
13:56:26 724  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 6, apiId = 0x26 0x05, full data = 0xfe 0x06 0x26 0x05 0x84 0x04 0x00 0x08 0x00 0x00 0xad, checksum = 0xad, error = false, errorMessage = null) 
13:56:26 729  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:26 730  DEBUG  Setting PAN to 1234.
13:56:26 730  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 4, apiId = 0x26 0x05, full data = 0xfe 0x04 0x26 0x05 0x83 0x02 0x34 0x12 0x80, checksum = 0x80, error = false, errorMessage = null) 
13:56:26 739  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:26 739  DEBUG  Setting Distribute Network Key to true.
13:56:26 740  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 3, apiId = 0x26 0x05, full data = 0xfe 0x03 0x26 0x05 0x63 0x01 0x00 0x42, checksum = 0x42, error = false, errorMessage = null) 
13:56:26 753  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:26 753  DEBUG  Setting Security Mode to 1.
13:56:26 753  DEBUG  -> ZB_WRITE_CONFIGURATION (Packet: length = 3, apiId = 0x26 0x05, full data = 0xfe 0x03 0x26 0x05 0x64 0x01 0x01 0x44, checksum = 0x44, error = false, errorMessage = null) 
13:56:26 764  DEBUG  <- ZB_WRITE_CONFIGURATION_RSP (ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)})
13:56:26 766  DEBUG  -> AF_REGISTER (Packet: length = 21, apiId = 0x24 0x00, full data = 0xfe 0x15 0x24 0x00 0x01 0x04 0x01 0x00 0x04 0x00 0x00 0x03 0x01 0x05 0x03 0x00 0x01 0x00 0x03 0x02 0x05 0x00 0x05 0x03 0x00 0x36, checksum = 0x36, error = false, errorMessage = null) 
13:56:26 775  DEBUG  <- AF_REGISTER_SRSP (AF_REGISTER_SRSP{Status=SUCCESS(0)})
13:56:26 776  DEBUG  Custom device 1024 registered at endpoint 1
13:56:26 776  DEBUG  Creating network as Coordinator
13:56:26 776  DEBUG  -> ZDO_MSG_CB_REGISTER (Packet: length = 2, apiId = 0x25 0x3e, full data = 0xfe 0x02 0x25 0x3e 0xff 0xff 0x19, checksum = 0x19, error = false, errorMessage = null) 
13:56:26 779  DEBUG  <- ZDO_MSG_CB_REGISTER_SRSP (ZDO_MSG_CB_REGISTER_SRSP{Status=SUCCESS(0)})
13:56:26 780  DEBUG  -> ZDO_STARTUP_FROM_APP (Packet: length = 2, apiId = 0x25 0x40, full data = 0xfe 0x02 0x25 0x40 0x00 0x00 0x67, checksum = 0x67, error = false, errorMessage = null) 
13:56:27 841  DEBUG  <- ZDO_STARTUP_FROM_APP_SRSP (ZDO_STARTUP_FROM_APP_SRSP{Status=FAILURE(1)})
13:56:27 841  INFO   Initialized ZigBee network with new or reset network state.
13:56:27 842  DEBUG  Waiting for network to become ready
13:56:27 846  DEBUG  <-- ZDO_STATE_CHANGE_IND (ZDO_STATE_CHANGE_IND{State=DEV_COORD_STARTING})
13:56:28 355  DEBUG  <-- ZDO_STATE_CHANGE_IND (ZDO_STATE_CHANGE_IND{State=DEV_COORD_STARTING})
13:56:28 866  DEBUG  <-- ZDO_STATE_CHANGE_IND (ZDO_STATE_CHANGE_IND{State=DEV_COORD_STARTING})
13:56:29 890  DEBUG  <-- ZDO_STATE_CHANGE_IND (ZDO_STATE_CHANGE_IND{State=DEV_ZB_COORD})
[... cut ...]
Node discovered: 00:12:4B:00:01:48:F1:20 (#0)
13:56:30 198  DEBUG  Endpoint inspection completed, next inspection slot in 0
13:56:30 199  DEBUG  Inspection queue: New queue size: 1. Failed queue size: 0
13:56:30 199  DEBUG  Inspecting device 00:12:4B:00:01:48:F1:20.
13:56:30 199  DEBUG  Inspecting existing node #0 (00:12:4B:00:01:48:F1:20)
13:56:30 199  DEBUG  Endpoint inspection completed, next inspection slot in 100
13:56:30 299  DEBUG  Inspection queue: New queue size: 0. Failed queue size: 0
```

Brhett, 2015-09-24:
Hi Cristiano,

I cannot reproduce it any more if I re-flash the dongle with the pre-built firmware **CC2531ZNP-Pro-Secure_LinkKeyJoin.hex**.

So the duplicate response of ***ZDO_SIMPLE_DESC_RSP*** should be caused by my previous self built one.

Regarding to the maximum numbers of clusters in per ***AF_REGISTER***, the "AppInClusterList" and "AppOutClusterList" might take up to 32 bytes according to the ZNP interfaces specification. So I think 16 clusters is the maximum number for each ***AF_REGISTER*** call.

Regards,
  Brhett

cdealti, 2015-09-24:
Hi Brhett,
  Thanks for the update. Glad to know it works now.
About the default endpoint issue, it does not make any sense to register such endpoint.
I'll open another issue for this.

Brhett, 2015-09-24:
@cdealti Thanks for your help.

The root cause is that I've built the firmware with "zgZdoDirectCB" set to TRUE in ZGlobals.c.

closed 2015-09-24 by Brhett