tlaukkan/zigbee4java #65: Discovery too chatty

cdealti, 2015-09-22:
I believe the discovery process is too chatty. In the log below (a network only composed by the dongle) look for occurrences of *Waiting for other request to complete*.
It looks like too many requests are issued concurrently and some are probably not needed.
Among the others:

At 11:09:39 698. ZDO_IEEE_ADDR_REQ is issued to get the IEEE address of the node and the nwk address of the associated nodes. This is actually not needed because, if the discovery starts from the coordinator, we already know its IEEE address (from ZB_GET_DEVICE_INFO_RSP). The list of associated nodes (with their IEEE addresses) can be derived from ZDO_MGMT_LQI_RSP.

So I don't see the need for ZDO_IEEE_ADDR_REQ.
What do you think?

```
11:09:39 697  DEBUG  Inspecting ZigBee network for new nodes.
11:09:39 698  DEBUG  -> ZDO_IEEE_ADDR_REQ (Packet: length = 4, apiId = 0x25 0x01, full data = 0xfe 0x04 0x25 0x01 0x00 0x00 0x01 0x00 0x21, checksum = 0x21, error = false, errorMessage = null) 
11:09:39 699  DEBUG  Inspecting ZigBee network for new nodes.
ZigBee API starting up ... [OK]
11:09:39 703  DEBUG  Inspection queue: New queue size: 0. Failed queue size: 0
There are 0 known devices in the network.
ZigBee console ready.
> 11:09:39 706  DEBUG  <- ZDO_IEEE_ADDR_REQ_SRSP (ZDO_IEEE_ADDR_REQ_SRSP{Status=SUCCESS(0)})
11:09:39 717  DEBUG  <-- ZDO_IEEE_ADDR_RSP (ZDO_IEEE_ADDR_RSP{AssocDevList=[], assocDevList=[], IEEEAddr=0x00 0x12 0x4b 0x00 0x01 0x48 0xf1 0x20, nwkAddr=0x00 0x00, NumAssocDev=0, SrcAddress=null, SrcAddrMode=0, StartIndex=0, Status=SUCCESS(0)})
11:09:39 718  DEBUG  -> ZDO_IEEE_ADDR_REQ (Packet: length = 4, apiId = 0x25 0x01, full data = 0xfe 0x04 0x25 0x01 0x00 0x00 0x00 0x00 0x20, checksum = 0x20, error = false, errorMessage = null) 
11:09:39 719  DEBUG  Waiting for other request to complete
11:09:39 725  DEBUG  <- ZDO_IEEE_ADDR_REQ_SRSP (ZDO_IEEE_ADDR_REQ_SRSP{Status=SUCCESS(0)})
11:09:39 726  DEBUG  -> ZB_GET_DEVICE_INFO (Packet: length = 1, apiId = 0x26 0x06, full data = 0xfe 0x01 0x26 0x06 0x06 0x27, checksum = 0x27, error = false, errorMessage = null) 
11:09:39 734  DEBUG  <-- ZDO_IEEE_ADDR_RSP (ZDO_IEEE_ADDR_RSP{AssocDevList=[], assocDevList=[], IEEEAddr=0x00 0x12 0x4b 0x00 0x01 0x48 0xf1 0x20, nwkAddr=0x00 0x00, NumAssocDev=0, SrcAddress=null, SrcAddrMode=0, StartIndex=0, Status=SUCCESS(0)})
11:09:39 735  DEBUG  ZDO_IEEE_ADDR_RSP from 0x00 0x12 0x4b 0x00 0x01 0x48 0xf1 0x20 with 0 associated
11:09:39 735  DEBUG  Waiting for other request to complete
11:09:39 736  DEBUG  <- ZB_GET_DEVICE_INFO_RSP (Packet: length = 9, apiId = 0x66 0x06, full data = 0xfe 0x09 0x66 0x06 0x06 0x34 0x12 0x00 0xff 0xfd 0x00 0x00 0x03 0x48, checksum = 0x48, error = false, errorMessage = null)
11:09:39 736  DEBUG  -> ZB_GET_DEVICE_INFO (Packet: length = 1, apiId = 0x26 0x06, full data = 0xfe 0x01 0x26 0x06 0x06 0x27, checksum = 0x27, error = false, errorMessage = null) 
11:09:39 739  DEBUG  Network browsing completed, waiting until 1442913879697
11:09:39 739  DEBUG  Inspecting device 00:12:4B:00:01:48:F1:20.
11:09:39 739  DEBUG  Waiting for other request to complete
11:09:39 741  DEBUG  <- ZB_GET_DEVICE_INFO_RSP (Packet: length = 9, apiId = 0x66 0x06, full data = 0xfe 0x09 0x66 0x06 0x06 0x34 0x12 0x00 0xff 0xfd 0x00 0x00 0x03 0x48, checksum = 0x48, error = false, errorMessage = null)
11:09:39 741  DEBUG  -> ZB_GET_DEVICE_INFO (Packet: length = 1, apiId = 0x26 0x06, full data = 0xfe 0x01 0x26 0x06 0x06 0x27, checksum = 0x27, error = false, errorMessage = null) 
11:09:39 741  DEBUG  ZDO_MGMT_LQI_REQ to 0 from index 0
11:09:39 741  DEBUG  Waiting for other request to complete
11:09:39 745  DEBUG  <- ZB_GET_DEVICE_INFO_RSP (Packet: length = 9, apiId = 0x66 0x06, full data = 0xfe 0x09 0x66 0x06 0x06 0x34 0x12 0x00 0xff 0xfd 0x00 0x00 0x03 0x48, checksum = 0x48, error = false, errorMessage = null)
11:09:39 746  DEBUG  -> ZDO_MGMT_LQI_REQ (Packet: length = 3, apiId = 0x25 0x31, full data = 0xfe 0x03 0x25 0x31 0x00 0x00 0x00 0x17, checksum = 0x17, error = false, errorMessage = null) 
11:09:39 746  DEBUG  Adding node #0 (00:12:4B:00:01:48:F1:20) to the network
Node added: 00:12:4B:00:01:48:F1:20 (#0)
11:09:39 746  DEBUG  Created node object for #0 (00:12:4B:00:01:48:F1:20) that was not available on the network
11:09:39 746  DEBUG  Inspecting new node #0 (00:12:4B:00:01:48:F1:20)
11:09:39 747  DEBUG  Waiting for other request to complete
11:09:39 756  DEBUG  <- ZDO_MGMT_LQI_REQ_SRSP (ZDO_MGMT_LQI_REQ_SRSP{Status=SUCCESS(0)})
11:09:39 756  DEBUG  -> ZDO_NODE_DESC_REQ (Packet: length = 4, apiId = 0x25 0x02, full data = 0xfe 0x04 0x25 0x02 0x00 0x00 0x00 0x00 0x23, checksum = 0x23, error = false, errorMessage = null) 
11:09:39 760  DEBUG  <-- ZDO_MGMT_LQI_RSP (ZDO_MGMT_LQI_RSP{NeighborLQICount=0, NeighborLQIEntries=0, NeighborLqiList=[], SrcAddress=0x00 0x00, StartIndex=0, Status=SUCCESS(0)})
11:09:39 760  DEBUG  Found 0 neighbors on node #0
```

tlaukkan, 2016-04-01:
I agree that the discovery would need some love. One additional thing would be to share list of already discovered nodes between all discovery modes.

tlaukkan, 2016-06-16:
Discovery has been re-implemented for Simple ZigBee API which is currently under experimental status.   For details see ZigBeeNetworkDiscoverer.

closed 2016-06-16 by tlaukkan