tlaukkan/zigbee4java #66: Registration of the default endpoint(s)

cdealti, 2015-09-24:
ApplicationFrameworkLayer.createDefaultSendingEndPoint() registers 3 endpoints, apparently without any real reason.

The code is weird too because it iterates over a set of 34 clusters and registers as many endpoints as needed (i.e. 3 endpoints) each with 16 clusters as inputs and outputs (the third only contains 2 clusters).

Also, each endpoint is registered specifying a device ID=0 in the Home Automation Profile which is the On/Off Switch in the HA specification.

I believe this default registration should not be hardcoded in the ApplicationFrameworkLayer and delegated to the application (e.g. the ZigBeeConsole).

Hint: the HA Combined Interface described in the HA spec could be a better choice.

Ciao,
  Cristiano

Brhett, 2015-09-24:
Hi Cristiano,

My understanding is that the default sending endpoint(s) implement all the clients to be able to interoperate with other servers of ZigBee devices. 
> I believe this default registration should not be hardcoded in the ApplicationFrameworkLayer and delegated to the application (e.g. the ZigBeeConsole).

Agree.

Regards,
Brhett

drugo72, 2015-09-28:
Hi Brhett,
  If you plan to create a PR with your fixes to addCustomEndopint, would you also consider the above?
Thanks,
  Cristiano

Brhett, 2015-09-29:
> The code is weird too because it iterates over a set of 34 clusters and registers as many endpoints as needed (i.e. 3 endpoints) each with 16 clusters as inputs and outputs (the third only contains 2 clusters).

Actually, the clusters are registered as outputs only. See https://github.com/tlaukkan/zigbee4java/blob/master/zigbee-api/src/main/java/org/bubblecloud/zigbee/network/impl/ApplicationFrameworkLayer.java#L202

I think its original purpose was create a ZigBee On/Off Switch to control the Philip Hue for example. But then it became to a "Combined Interface Device"+. It is used to control other ZigBee devices as many as possible.

In order to be HA specification compliant, I think there should be several specific endpoints being registered to cover the functionalities. e.g. "Combined Interface Device" + "Thermostat" + "Home Gateway" + "IAS CIE" + ... However, some of them have common clusters, which might introduce more complexity.

So I think it's better to delegate the endpoints registration to the application. e.g. ZigBeeConsole