tlaukkan/zigbee4java #67: Deserialization of network persistence failed

cdealti, 2015-09-24:
This is probably well known.
Ciao,
  Cristiano

```
ZigBee API starting up...java.lang.RuntimeException: Error serializing network state.
	at org.bubblecloud.zigbee.network.impl.NetworkStateSerializer.deserialize(NetworkStateSerializer.java:77)
	at org.bubblecloud.zigbee.ZigBeeApi.deserializeNetworkState(ZigBeeApi.java:353)
	at org.bubblecloud.zigbee.ZigBeeConsole.start(ZigBeeConsole.java:110)
	at org.bubblecloud.zigbee.ZigBeeConsoleJavaSE.main(ZigBeeConsoleJavaSE.java:34)
Caused by: org.codehaus.jackson.map.JsonMappingException: No suitable constructor found for type [simple type, class org.bubblecloud.zigbee.network.ZigBeeNodeDescriptor]: can not instantiate from JSON object (need to add/enable type information?)
 at [Source: java.io.StringReader@26eb6935; line: 1, column: 289] (through reference chain: org.bubblecloud.zigbee.network.impl.ZigBeeEndpointImpl["node"]->org.bubblecloud.zigbee.network.impl.ZigBeeNodeImpl["nodeDescriptor"])
	at org.codehaus.jackson.map.JsonMappingException.from(JsonMappingException.java:163)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserializeFromObjectUsingNonDefault(BeanDeserializer.java:740)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:683)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserialize(BeanDeserializer.java:580)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:87)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromObject(AsArrayTypeDeserializer.java:55)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserializeWithType(BeanDeserializer.java:664)
	at org.codehaus.jackson.map.deser.SettableBeanProperty.deserialize(SettableBeanProperty.java:297)
	at org.codehaus.jackson.map.deser.SettableBeanProperty$MethodProperty.deserializeAndSet(SettableBeanProperty.java:414)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:697)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserialize(BeanDeserializer.java:580)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:87)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromAny(AsArrayTypeDeserializer.java:69)
	at org.codehaus.jackson.map.deser.AbstractDeserializer.deserializeWithType(AbstractDeserializer.java:77)
	at org.codehaus.jackson.map.deser.SettableBeanProperty.deserialize(SettableBeanProperty.java:297)
	at org.codehaus.jackson.map.deser.SettableBeanProperty$MethodProperty.deserializeAndSet(SettableBeanProperty.java:414)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:697)
	at org.codehaus.jackson.map.deser.BeanDeserializer.deserialize(BeanDeserializer.java:580)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:87)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromAny(AsArrayTypeDeserializer.java:69)
	at org.codehaus.jackson.map.deser.std.UntypedObjectDeserializer.deserializeWithType(UntypedObjectDeserializer.java:106)
	at org.codehaus.jackson.map.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:219)
	at org.codehaus.jackson.map.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:194)
	at org.codehaus.jackson.map.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:30)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer._deserialize(AsArrayTypeDeserializer.java:87)
	at org.codehaus.jackson.map.jsontype.impl.AsArrayTypeDeserializer.deserializeTypedFromArray(AsArrayTypeDeserializer.java:45)
	at org.codehaus.jackson.map.deser.std.CollectionDeserializer.deserializeWithType(CollectionDeserializer.java:232)
	at org.codehaus.jackson.map.deser.StdDeserializerProvider$WrappedDeserializer.deserialize(StdDeserializerProvider.java:494)
	at org.codehaus.jackson.map.ObjectMapper._readMapAndClose(ObjectMapper.java:2723)
	at org.codehaus.jackson.map.ObjectMapper.readValue(ObjectMapper.java:1854)
	at org.bubblecloud.zigbee.network.impl.NetworkStateSerializer.deserialize(NetworkStateSerializer.java:75)
	... 3 more

```

Brhett, 2015-09-28:
Could be resolved by https://github.com/Brhett/zigbee4java/commit/14cc813b1199eacf02618b32a1586d9bf1c272a3

drugo72, 2015-09-28:
Good! thanks.

PR #69 (2015-10-02): Bugfix
Fix #48 
Fix #67
commit 14cc813b (2015-10-02): Fix #67: add default constructors for ZigBeeNodeDescriptor and ZigBeeNodePowerDescriptor to resolve the deserialization issue
closed 2015-10-02 by cdjackson