tlaukkan/zigbee4java #73: Zigbee console join enable / disable reports error

tlaukkan, 2016-01-04:
Hi

I am observing with master HEAD that although ZDO_MGMT_PERMIT_JOIN_REQ passes successfully an error is reported related to it both at startup and when running join command from console. I wonder if there is some problem with handling of the commands directed to the zigbee dongle or error handling on console level? These commands do not access anything over network so their status is returned synchronously via ZDO_MGMT_PERMIT_JOIN_REQ_SRSP. It also looks like the ZDO_MGMT_PERMIT_JOIN_REQ is sent twice.

Despite this I was able to join philips hue bulb to the network and command it.

Br,
Tommi

ZigBee API starting up...20:20:21 735  INFO   Initialized ZigBee network with existing network state.
20:20:21 799  INFO   Registered default sending endpoint 1 with clusters: [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 15, 10, 12, 21, 1794, 256]
20:20:21 823  INFO   Registered default sending endpoint 2 with clusters: [257, 258, 512, 513, 514, 516, 768, 1027, 1024, 1025, 1026, 1028, 1029, 1030, 2820, 1280]
20:20:21 842  INFO   Registered default sending endpoint 3 with clusters: [1281, 1282]
20:20:29 890  ERROR  Error sending ZDO_MGMT_PERMIT_JOIN_REQ
ZigBee API starting up ... [OK]
There are 1 known devices in the network.
ZigBee console ready.

> join enable
20:17:09 450  DEBUG  Sending permit join with data: -1
20:17:09 451  DEBUG  -> ZDO_MGMT_PERMIT_JOIN_REQ (Packet: length = 5, apiId = 0x25 0x36, full data = 0xfe 0x05 0x25 0x36 0x0f 0xfc 0xff 0xff 0x01 0xe4, checksum = 0xe4, error = false, errorMessage = null)
20:17:09 468  DEBUG  <- ZDO_MGMT_PERMIT_JOIN_REQ_SRSP (ZDO_MGMT_PERMIT_JOIN_REQ_SRSP{Status=SUCCESS(0)})
20:17:09 470  DEBUG  -> ZDO_MGMT_PERMIT_JOIN_REQ (Packet: length = 5, apiId = 0x25 0x36, full data = 0xfe 0x05 0x25 0x36 0x02 0x00 0x00 0xff 0x01 0xea, checksum = 0xea, error = false, errorMessage = null)
20:17:09 488  DEBUG  <- ZDO_MGMT_PERMIT_JOIN_REQ_SRSP (ZDO_MGMT_PERMIT_JOIN_REQ_SRSP{Status=SUCCESS(0)})
20:17:17 489  ERROR  Error sending ZDO_MGMT_PERMIT_JOIN_REQ
ZigBee API permit join enable ... [FAIL]

Brhett, 2016-01-05:
https://github.com/tlaukkan/zigbee4java/commit/0fbd31beba44de1ecb8d15ada1aa0a655db720e9 sent permit join request to routers and coordinators separately.

And this issue cannot be reproduced on my side (master head + CC2531 only).

cdealti, 2016-01-07:
No problem for me too (master head + CC2531 only). I'm on Linux though.

commit c35341aa (2016-03-30): Ignored permit join response as local coordinator does not send it if network is empty. Fixed #73.
closed 2016-03-30 by tlaukkan