tlaukkan/zigbee4java #75: sendSynchrouns(hwDriver, request, RESEND_TIMEOUT) = null  in windows 10

SamLuky, 2016-01-07:
Hello :) 

i  tried to run zig4java console with eclipse in windows 10 but the application crash even if all the other configurations are well ( dongle, firmware, args ). 

the crash occur when the step of creation of zigbee network is begin ! the reason is : sendSynchrouns(hwDriver, request, RESEND_TIMEOUT) = null  

whereas when i  run the same program in eclipse using a ubuttu machines the program run well and the console of the application is running .  the value of sendSynchrouns(hwDriver, request, RESEND_TIMEOUT) in this case is : 

sendSynchrouns(hwDriver, request, RESEND_TIMEOUT) = ZB_WRITE_CONFIGURATION_RSP{Status=SUCCESS(0)} 

hope that helps someone :)

cdealti, 2016-01-10:
Can you set the log level in ```./zigbee-api/src/main/resources/log4j.xml``` to ```trace``` and provide a log?

SamLuky, 2016-01-11:
hello ! i find that the problem was in the system Windows 10 ! i put the same program in Ubuntu and it's run well :)

cdealti, 2016-01-11:
It should work on windows 10 too. However there have been reported other issues on Windows 10 (see #74).

tlaukkan, 2016-04-01:
Hi Sam, do you still experience this problem? I am running without problems in windows 10 at the moment.

commit 45fc5f70 (2016-04-04): Implemented ZigbeeSerialPort with jSSC. Switched default port implementation to be jSSC. Fixes #75 and fixes #79.
closed 2016-04-04 by tlaukkan