tlaukkan/zigbee4java #78: Help needed

pascalleclercq, 2016-02-09:
Hi, 
I'm a newbie in IoT and in Zigbee, so sorry if I ask too obvious questions.

I'm trying to use zigbee4java based on this sample : https://github.com/jpizarrom/kura-greenhouse-demo-zigbee
I use this hardware : http://www.embit.eu/products/wireless-modules/emb-z2530pa/

So far I don't think It works and I have strange error message :  
ERROR o.b.z.n.p.ZigBeeNetworkManagerImpl - Unable to reset dongle

Not sure what It means but I don't think my zigbee device is about to work properly.

Does anybody can tell me what's going wrong ?

Thanks in advance.

Here is a more complete log : 

2007-01-01 01:29:51,428 [pool-14-thread-1] TRACE o.b.z.n.p.ZToolPacketParser - Creating ZToolPacketParser
2007-01-01 01:29:51,431 [pool-14-thread-1] TRACE o.b.z.n.p.ZToolPacketParser - Provided InputStream class com.smarthome.gateway.zigbee.LoggingInputStream doesn't provide the mark()/reset() feature, wrapping it up as BufferedInputStream
2007-01-01 01:29:51,441 [pool-14-thread-1] TRACE o.b.z.n.p.ZigBeeNetworkManagerImpl$WaitForCommand - Waiting for asynchronous response message 16768.
2007-01-01 01:29:51,444 [ZToolPacketParser] TRACE o.b.z.n.p.ZToolPacketParser - ZToolPacketParser parserThread started
2007-01-01 01:29:51,473 [pool-14-thread-1] DEBUG o.b.z.n.p.ZigBeeInterface - -> SYS_RESET (Packet: length = 1, apiId = 0x41 0x00, full data = 0xfe 0x01 0x41 0x00 0x01 0x41, checksum = 0x41, error = false, errorMessage = null) 
2007-01-01 01:29:51,477 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "[0xfe]"
2007-01-01 01:29:51,480 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "[0x1]"
2007-01-01 01:29:51,482 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "A"
2007-01-01 01:29:51,484 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "[0x0]"
2007-01-01 01:29:51,487 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "[0x1]"
2007-01-01 01:29:51,489 [pool-14-thread-1] INFO  c.s.g.z.Wire -  >> "A"
2007-01-01 01:29:56,493 [pool-14-thread-1] TRACE o.b.z.n.p.ZigBeeNetworkManagerImpl$WaitForCommand - Timeout 5000 expired and no packet with 16768 received
2007-01-01 01:29:56,496 [pool-14-thread-1] ERROR o.b.z.n.p.ZigBeeNetworkManagerImpl - Unable to reset dongle
2007-01-01 01:29:56,504 [pool-14-thread-1] TRACE o.b.z.n.p.ZigBeeNetworkManagerImpl - HARDWARE_INITIALIZING -> CLOSED
2007-01-01 01:29:56,507 [pool-14-thread-1] INFO  c.s.g.z.ZigBeeDemo - post initializeHardware false
**2007-01-01 01:29:56,510 [pool-14-thread-1] INFO  o.b.z.n.p.ZigBeeNetworkManagerImpl - Failed to reach the NETWORK_READY level: getCurrentChannel() failed
2007-01-01 01:29:56,512 [pool-14-thread-1] INFO  o.b.z.n.p.ZigBeeNetworkManagerImpl - Failed to reach the NETWORK_READY level: getCurrentPanId() failed**
2007-01-01 01:29:56,515 [pool-14-thread-1] INFO  c.s.g.z.ZigBeeDemo - ZigBee current pan=-1, channel=-1. Network will be reset.

cdealti, 2016-02-09:
Hi Pascal,
  I'm a Kura commiter working at Eurotech. We (Eurotech) plan to integrate zigbee4java in the next Kura release.

Can you please try building zigbee4java from my [fork] (https://github.com/cdealti/zigbee4java/tree/pr-fix-issue-76)? It contains the most recent fixes that have not been yet merged upstream.

However, before doing this you need to make sure the Embit dongle is loaded with a suitable ZNP TI [firmware](https://github.com/tlaukkan/zigbee4java), see the **README**.

The problem with Embit module is that it has a power amplifier and this usually requires recompiling the firmware after configuring the PA setting. Please refer to the TI Z-Stack and the Embit documentation for information on what settings to change. They usually provide good documentation with all the information needed.

We also plan to use an Embit module but based on the CC2538. I have the DK but no time to give it a try at the moment.

pascalleclercq, 2016-02-09:
Thank for you feedback @cdealti ! I try your fix and I get back to you asap.

pascalleclercq, 2016-02-10:
Hi @cdealti 
Just to inform you, I update ly zigbee4java with your jars but not the firmware.
I still have the "ERROR o.b.z.n.p.ZigBeeNetworkManagerImpl - Unable to reset dongle".

Will try to recompile the firmware as recommended but It will take more time....

Thanks again.

tlaukkan, 2016-06-16:
Please use firmware suggested in readme file and latest release or master head.

closed 2016-06-16 by tlaukkan