tlaukkan/zigbee4java #91: "quit" commad can't quit the console application

Brhett, 2016-06-08:
The "read" method is blocking in the thread "ZToolPacketParser". Thus, "parserThread.join()" might be blocked during closing of "ZToolPacketParser".

Tested on the current head version eaf7129.

tlaukkan, 2016-06-09:
Repeats still on Macbook Pro.

Brhett, 2016-06-13:
The root cause should be the port implementation of libjSSC-2.8_[x86].so that the `select()` is blocked even if the handle was closed.
A possible trick to fix this issue is adding a timeout to `select()` and check the return value of it. See pseudo-code below.

```
int ret = select(portHandle +１, &read_fd_set, NULL, NULL, &internal_timeout);
if (ret == -1)
{
    throwSerialPortException(...);
    return NULL;
}
else if (!FD_ISSET(portHandle, &read_fd_set))
{
    continue;
}
```

commit fba00938 (2016-06-08): Fixes #91 by setting packet parser thread to be daemon thread. Please reopen ticket if still occurs. This bug does not reproduce on Windows 10 platform.