tlaukkan/zigbee4java #103: Code styling

cdjackson, 2016-08-29:
It seems you enabled a CI tonight for code formatting and it has pretty tight requirements on how the code is styled. I've spent a lot of time tonight trying to get this correct as I've had well over 100 emails from the CI bot. Please can you either provide information on how you want it styled or disable this. It could easily spend days guessing what this thing wants.

I tried googling for a formatter file (I'm using Eclipse) but couldn't find anything. I can't believe there's not something available given how picky the bot is. I've tried using a formatter for another project I work on that has tight requirements, and that made things worse - sorry!

I think the only way this will work is if there is an auto formatter somewhere - either in the editor, or as part of the CI (which probably doesn't work as it would need to commit to git).

cdjackson, 2016-08-31:
@tlaukkan any chance of disabling the code review please. Otherwise I'll close all PRs until this is sorted? When I first created these PRs over the weekend the spam generator didn't do anything, so I don't know if you enabled it earlier in the week?

tlaukkan, 2016-12-04:
Hi. I am sorry about the inconvenience. I think the style checks had been on for  a while. I do not know why it suddenly activated for you. In general I think it is useful to have it when changes are smaller but on larger refactoring it does spam quite much. The code style requirements are google Java code style conventions and you are right, we should include definition file to the project.