tlaukkan/zigbee4java #107: Hue Tap support

TheNetStriker, 2016-11-16:
I would like to react to Hue Tap switches using this library, but so far I didn't found out if this is possible. I've logged the packets my Hue Tap Switch is sending and uploaded a screenshot of the packets to my Dropbox: [Link](https://www.dropbox.com/s/yv8sxldidmn7ffd/HueTap.png?dl=0)

Every button click sends three packets. In the screenshots are the packets of button 1, 2, 3 and 4. The Hue Tap switch sends all packets to the 0xFFFF pan id and the packets are not encrypted.

Is there a way to react to those packets using this library?

tlaukkan, 2016-12-04:
Hi, are the tap switches compatible with ZigBee Home Automation Profile and are you able to join them to network? If they do not join the network then as far as I know it is not possible to use them. If they join and send commands according to ZigBee HA profile then they can be utilized using this library.

TheNetStriker, 2016-12-04:
I read somewhere that the Tap Switches are "ZigBee Green Power Stack" devices, but I'am no expert regarding ZigBee. I wasn't able to join them, but I was able to take over my whole existing Philipps Hue with the help of the guy from this [blog](http://peeveeone.com/?p=166).

He found the key Hue is using to exchange the network key when joining new lamps. With this key and a sniffer I was able to find out the network key from my Hue system and was able to decode all ZigBee packets. After that I simply programmed the ZigBee usb stick to my existing network id and network key and started up your server example. My hue lights where automatically added when turning them on. (Also the older ones that doesn't support network join) But nothing happened when pressing one of the Tap buttons.

I read somewhere that Texas Instruments is trying to implement the Green Stack into the usb sticks firmware. Do you now something about that?

TheNetStriker, 2017-02-09:
I just noticed that version 3.0.0 of Z-Stack was released where "support Green Power Proxy" is mentioned: http://www.ti.com/tool/Z-STACK

Is this maybe the key in getting the HUE Tap events?