tlaukkan/zigbee4java #109: Error in zigbee api ( reporting command)

han-alink, 2016-11-28:
MinReporting interval is set twice (instead of MIN and MAX

        record.setMinimumReportingInterval(minInterval);
        record.setMinimumReportingInterval(maxInterval);
        record.setReportableChange(reportableChange);

tlaukkan, 2016-11-29:
Thank you Han for the bug reports. I try to look at this during the weekend. If you have pull requests those would be much appreciated. I have some other pull requests in queue to check as well.

han-alink, 2016-11-30:
Hi Tommy,

Glad to hear that you are still actively working on the zigbee4java project.

I’m not familiar with “pull requests”. I imported zigbee4java in Eclipse and are using the software in combination with HomeRun (my home control program). I’m not yet finished with it yet, but can control my Zigbee devices already and get reports from the sensor devices. I added the “LevelContrallableOutput device” (deviceId = 3). I think you should rename the device with deviceId = 1 to LevelControlSwitch 

I attached a patch file. This also contains some debugging statements which you should ignore. Have a close look and be warned: I’m not an expirienced java programmer!

Thanks for your effort,

Bset regards,

Han

commit 5f5e2ac0 (2016-12-04): Fixes #109
closed 2016-12-04 by tlaukkan