tlaukkan/zigbee4java #111: Xiaomi Aqara temperature humidity sensor

DreyerAc, 2019-02-20:
Hello,

I´m using your libary to communicate with some develco smart plugs. Now I want to add a Xiaomi Aqara temperature humidity sensor. 
But if I try to join the device to the zigbee network the following exception is thrown:

java.lang.IllegalArgumentException: No sender EndPoint 1 with cluster 1026.profiling.org.bubblecloud.zigbee.network.packet.af.AF_INCOMING_MSG - 2019-02-19 10:46:38 - DEBUG - AF_INCOMING_MSG: creating object
profiling.org.bubblecloud.zigbee.network.packet.af.AF_INCOMING_MSG - 2019-02-19 10:46:38 - DEBUG - AF_INCOMING_MSG: object created
org.bubblecloud.zigbee.network.impl.CommandInterfaceImpl - 2019-02-19 10:46:38 - DEBUG - <-- AF_INCOMING_MSG ([I@393ce184)
org.bubblecloud.zigbee.v3.zcl.ZclCommandTransmitter - 2019-02-19 10:46:38 - DEBUG - Received command: [ clusterId: 1029 commandId: 10 specific: false extension: false ZCL Header: 0x18 0x07 0x0a, ZCL Payload: 0x00 0x00 0x21 0xa4 0x1d]
org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerImpl.AFMessageListenerFilter - 2019-02-19 10:46:38 - ERROR - Error AF message listener notify.

	at org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer.getSenderEndpointProfileId(ApplicationFrameworkLayer.java:136) 

Is there any support for Xiaomi devices for your library?

Thanks in advance
Best Regards
Achim

AndrejFusek, 2019-02-20:
Hello DreyerAc,
I think You need my fork <https://github.com/AndrejFusek/zigbee4java> of
the zigbee4java library. I added support for some of Xiaomi smart devices
;) It should work with Xiaomi temperature and humidity sensor and also
smart socket (I tested the library with these 2 devices). It looks like I
did it just for You :D
Good luck (and star my fork of zigbee4java if it will work ;) )

Andy

DreyerAc, 2019-02-21:
Hello Andy,

thanks for your hint. I´ve tried your fork with the console app. The nodes are added to the network. But they are discovered as a Color Dimmable Light and the correct attributes are not available. Here are the parts from my console log:

1. Discovery:
> join enable
ZigBee API permit join enable ... [OK]
 [ZToolPacketParser] INFO org.bubblecloud.zigbee.network.discovery.AnnounceListenerImpl - Device announcement received - Network Address: #823, IEEE Address: 00:15:8D:00:02:54:D2:CF
Node added: 00:15:8D:00:02:54:D2:CF (#823)[EndpointBuilder[org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerImpl@13deb50e]] WARN org.bubblecloud.zigbee.network.discovery.EndpointBuilder - Created node object for #823 (00:15:8D:00:02:54:D2:CF) that was not available on the network
[ZToolPacketParser] INFO org.bubblecloud.zigbee.network.discovery.AnnounceListenerImpl - Device announcement received - Network Address: #823, IEEE Address: 00:15:8D:00:02:54:D2:CF
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.
Device added: 00:15:8D:00:02:54:D2:CF/1 (#823)
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.
Device added: 00:15:8D:00:02:54:D2:CF/2 (#823)
Device added: 00:15:8D:00:02:54:D2:CF/3 (#823)
Node discovered: 00:15:8D:00:02:54:D2:CF (#823)

2. List Command:
> list
0) 00:15:8D:00:02:54:D2:CF/1 [823] : Color Dimmable Light
1) 00:15:8D:00:02:54:D2:CF/2 [823] : Color Dimmable Light
2) 00:15:8D:00:02:54:D2:CF/3 [823] : Color Dimmable Light

3. Describe of the 3 Devices:
> desc 0

Network Address  : 823

> 
Extended Address : 00:15:8D:00:02:54:D2:CF

> 
Endpoint Address : 1

> 
Device Profile   : ZigBee Home Automation  (0x0104)

> 
Device Category  : null

> 
Device Type      : Color Dimmable Light  (0x5F01)

> 
Device Version   : 1

> 
Implementation   : org.bubblecloud.zigbee.api.device.impl.ColorDimmableLightDevice

> 
Input Clusters   : 

> 
                 : -1 null

> 
                 : 0 Basic

> 
                 :    0 r-- ZCLVersion   [UnsignedInteger8bit]

> 
                 :    1 r-- ApplicationVersion   [UnsignedInteger8bit]

> 
                 :    2 r-- StackVersion   [UnsignedInteger8bit]
[... 275 lines cut ...]
                 :    85 rws PresentValue (0)  [SinglePrecision]

> 
                 :    103 rw- Reliability   [Enumeration8bit]

> 
                 :    106 rw- Resolution   [SinglePrecision]

> 
                 :    111 r-s StatusFlags (0)  [Bitmap8bit]

> 
                 :    117 rw- EngineeringUnits   [Enumeration16bit]

> 
                 :    256 rw- ApplicationType   [UnsignedInteger32bit]

> 
>

Do you have any idea whats the problem?

Thanks for you help
Regards
Achim

AndrejFusek, 2019-02-22:
>    1. List Command:
>    1. 00:15:8D:00:02:54:D2:CF/1 [823] : Color Dimmable Light
>    2. 00:15:8D:00:02:54:D2:CF/2 [823] : Color Dimmable Light
>    3. 00:15:8D:00:02:54:D2:CF/3 [823] : Color Dimmable Light
>    1. Describe of the 3 Devices:

It is nice to see that the device was successfully added :) but what to do
with the problem... Which command do You use for device list? Please try
"help" command and send me the printed list of possible commands please,
because I forgot almost all commands. It is more than a year when I was
working with these commands through the console.

DreyerAc, 2019-02-24:
No problem. I tought to test with the console is the simplest way.

These are the available commands:

> help

Commands:

> 
bind - Binds a device to another device.

> 
color - Changes light color.

> 
desc - Describes a device.

> 
enroll - Enrolls IAS Zone device to this CIE device by setting own address as CIE address to the
 IAS Zone device.

> 
help - View command help.

> 
join - Enable or diable network join.

> 
level - Changes device level for example lamp brightness, where LEVEL is between 0 and 1.

> 
list - Lists devices.

> 
listen - Listen to attribute reports.

> 
lock - Locks door.

> 
lqi - List LQI neighbours list.

> 
nodes - Lists physical node information.

> 
off - Switches device off.

> 
on - Switches device on.

> 
quit - Quits console.

> 
read - Read an attribute.

> 
squawk - Make the specfied IAS warning device squawk.

> 
subscribe - Subscribe to attribute reports.

> 
unbind - Unbinds a device from another device.

> 
unlisten - Unlisten from attribute reports.

> 
unlock - Unlocks door.

> 
unsubscribe - Unsubscribe from attribute reports.

> 
warn - Warn IAS warning device.

> 
write - Write an attribute.

AndrejFusek, 2019-02-24:
Yes, via console is the simplest way, but I've been developing my own
application which uses this library, so this is the reason why I forgot
thing about the console :)
Okay, so did You use a command *join*, then add your temperature sensor by
holding its button for a while, and then wait for the library to add the
device, and then You use a command *list *to see list of devices? Please
send me also the output of the *list *command.

DreyerAc, 2019-02-26:
Yes. I´ve joined the device as you describe above.
The output of the list command is the following:

> list

0) 00:15:8D:00:02:54:D2:CF/1 [12353] : Color Dimmable Light

> 
1) 00:15:8D:00:02:54:D2:CF/2 [12353] : Color Dimmable Light

> 
2) 00:15:8D:00:02:54:D2:CF/3 [12353] : Color Dimmable Light

>

AndrejFusek, 2019-02-26:
>    1. 00:15:8D:00:02:54:D2:CF/1 [12353] : Color Dimmable Light
>    1. 00:15:8D:00:02:54:D2:CF/3 [12353] : Color Dimmable Light

If I remember correctly, these 3 devices are just for Zigbee USB Adapter
(like server). If You pair one device (like your temperature sensor), there
should be 4 devices.
It looks like Your sensor is not added at all. I looked at the log from
Your second email and noticed something:
WARN org.bubblecloud.zigbee.network.discovery.EndpointBuilder - Created
node object for #823 (00:15:8D:00:02:54:D2:CF) *that was not available on
the network*
Are You sure You are using current master (commit a1bd53b
<https://github.com/AndrejFusek/zigbee4java/commit/a1bd53bd690564712561deaff987cf4e8d0bdfcc>)
of https://github.com/AndrejFusek/zigbee4java  ?

DreyerAc, 2019-02-26:
Yes your are right. There are some error messages:

> [EndpointBuilder[org.bubblecloud.zigbee.network.impl.ZigBeeNetworkManagerImpl@13deb50e]] WARN org.bubblecloud.zigbee.network.discovery.EndpointBuilder - Created node object for #12353 (00:15:8D:00:02:54:D2:CF) that was not available on the network
[ZToolPacketParser] INFO org.bubblecloud.zigbee.network.discovery.AnnounceListenerImpl - Device announcement received - Network Address: #12353, IEEE Address: 00:15:8D:00:02:54:D2:CF
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.

Yes I´m using the master branch of your repository.

AndrejFusek, 2019-02-26:
Hmmm, I remember some of these messages.
"Created node object for #12353 (...) that was not available on the
network" - this might be the problem, I will try to remember how I solved
it in the past...
"No sender EndPoint 1 with cluster 1026, so found in EndPoint 2" - OK, this
message was written by me. It is WARN that this is just workaround to make
work some devices like temperature sensors.

One another idea: please run *nodes* command and send me the output.
And did You try to change temperature of the sensor after pairing (move
some other room/to fridge)? *Maybe *it can send notifications about
temperature changes without being in the device list, but I am not sure -
just try it please.

DreyerAc, 2019-02-26:
First of all the output of the nodes command:

> nodes

IEEE Address     : 00:12:4B:00:19:36:75:8B
Network Address  : #0
Node Descriptor  : Logical Type       COORDINATOR
                 : Manufacturer Code  00A0
                 : Max Buffer Size    1
                 : Max Transfer Size  40960
                 : MAC Capabilities   [SECURITY_CAPABLE]
                 : Server Mask        []
Power Descriptor : Power Mode         RECEIVER_ON
                 : Power Source       MAINS
                 : Power Level        FULL
                 : Power Available    [MAINS]

IEEE Address     : 00:15:8D:00:02:54:D2:CF
Network Address  : #44814
Node Descriptor  : Logical Type       END_DEVICE
                 : Manufacturer Code  0064
                 : Max Buffer Size    0
                 : Max Transfer Size  25600
                 : MAC Capabilities   [PAN_COORDINATOR, FFD, MAINS_POWER, RX_ALWAYS_ON, SECURITY_CAPABLE]
                 : Server Mask        []

Power Descriptor : Power Mode         RECEIVER_ON
                 : Power Source       BATTERY_DISPOSABLE
                 : Power Level        FULL
                 : Power Available    [BATTERY_DISPOSABLE]

Second I put the sensor after pairing on my window board outside. ;-) But there is only the following repeating output:
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1026, so found in EndPoint 2.
[ZToolPacketParser] WARN org.bubblecloud.zigbee.network.impl.ApplicationFrameworkLayer - No sender EndPoint 1 with cluster 1029, so found in EndPoint 2.

AndrejFusek, 2019-03-03:
Hi again, sorry for my latency, I was really busy.
I remembered that when I paired my Xiaomi temperature & humidity sensor, it immediately sends information about temperature and humidity (automatically) and also when one of the values was changed. No type of subscribing was needed.
Do You have that Xiaomi sensor with shape of circle or rounded square? I had circle-shaped and that worked. I am afraid Your is rounded square-shape and this type might have different codes in Zigbee messages or something... If so, I can't help You, You can try to find something useful on the internet to compare these two types of the sensor.